adafruit/tinyuf2 #1: Factory reset option

chegewara, 2020-06-21:
There is another way to restart from uf2 partition. IDF is using something called factory reset, it can be enabled in menuconfig and is using selected pin to erase ota_data partition and boot from factory partition. I think this is better than:
> GPIO is attached to an 100K resistor and 1uF Capacitor to serve as 1-bit memory, which hold the pin value long enough for double reset detection

Thanks for this project.

PS is there a way or tool to build UF2 partition?

hathach, 2020-06-23:
> There is another way to restart from uf2 partition. IDF is using something called factory reset, it can be enabled in menuconfig and is using selected pin to erase ota_data partition and boot from factory partition. I think this is better than:

Thanks, we are aware of that option, but it requires an additional GPIO which is what we are trying to reduce now.

closed 2020-06-23 by hathach

chegewara, 2020-06-23:
> Thanks, we are aware of that option, but it requires an additional GPIO which is what we are trying to reduce now.

I dont know all about that project, but as far as i understand it is about using native USB on esp32 S2, which is using pins 19 and 20. Is there any reason to not use them as factory reset pins?