adafruit/tinyuf2 #5: Finalize partition layout

tannewt, 2020-08-26:
Hey, could we finalize the partition layout? I've added native WiFi to the CircuitPython build and I'm over `0x130000` bytes per partition. I'm thinking we'll want to do 1MB for the filesystem and then split the rest between two OTA and UF2. We aren't using NVS at all. What size do we want to leave for UF2?

It'd be great to share the partition layout with Arduino as well. @me-no-dev what partition layout are you using for Arduino?

hathach, 2020-08-27:
currently the bootloader take 200 KB, I will try to see if I could optimize it, it seems like lto is not enabled for IDF. I will try to see if I could shrink it down furthermore. 

```
            Archive File DRAM .data & .bss & other   IRAM   D/IRAM Flash code & rodata   Total
                  libc.a        364      4       0      0      368      54973     3794   59503
                   (exe)         92   5281       0      0     5376      20749      476   31974
               libmain.a        417  10697       0      0    11114       1869     1091   25188
           libfreertos.a       2080    736       0      0    12663        133     1386   16998
                libsoc.a       1058      0       0      0     9260       3305      947   14570
             libdriver.a        124     66       0      0     1780       7362     2137   11469
          libspi_flash.a       1053    416       0      0     7245       1358     1228   11300
         libesp_system.a       2363     21       0      0     5806        785      792    9767
         libesp_common.a        153     84       0      0      452       1375     5862    7926
            libesp32s2.a         16     21       0      0     1246       2979     1537    5799
                libvfs.a        272     48       0      0      320       4389      560    5589
               libheap.a        301      8       0      0     3003        830      729    4871
 libbootloader_support.a          0      4       0      0        4       2717     1715    4440
        libesp_ringbuf.a          0      0       0      0     2934          0      411    3345
             libnewlib.a        190    264       0      0     1367        724       91    2636
         libmbedcrypto.a          0      0       0      0        0       1459      365    1824
         libapp_update.a          1     12       0      0      167        612      608    1400
                liblog.a          8    272       0      0      486        383      150    1299
          libesp_timer.a         16     20       0      0      332        607      226    1201
              libefuse.a         28      4       0      0       32        720      394    1178
          libled_strip.a          0     16       0      0      181        597      377    1171
             libxtensa.a          0      0       0      0      409          0       58     467
             libboards.a          0      9       0      0        9        367       20     405
                libhal.a          0      0       0      0      333          0       32     365
            libpthread.a          8     12       0      0       20        246        0     286
        libsoc_esp32s2.a          0      0       0      0        0          0      192     192
                libgcc.a          0      0       0      0        0          0      160     160
            libesp_rom.a          0      0       0      0       24          0        0      24
                libcxx.a          0      0       0      0        0         11        0      11
```

me-no-dev, 2020-08-27:
@tannewt wee have different partition schemes available in Arduino. Depending on size of apps, storage and flash in general. Apps that include both BT and WiFi easily go over 1MB (even 1.5MB).
https://github.com/espressif/arduino-esp32/tree/master/tools/partitions

tannewt, 2020-08-27:
@hathach My understanding is that LTO is off because it can mess up memory sections.

@me-no-dev Thanks for the link! It'd be great to have one compatible Arduino partition.

tannewt, 2020-08-27:
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default_ffat.csv and https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/ffat.csv look the closest to what we have.

tannewt, 2020-08-27:
Ok, I've updated [the CircuitPython partition layout](https://github.com/tannewt/circuitpython/blob/native_wifi/ports/esp32s2/partitions.csv) in my `native_wifi` branch to:

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
# bootloader.bin          0x1000
# partition table         0x8000, 0xC00
otadata,  data, ota,      0xd000,  0x2000,
phy_init, data, phy,      0xf000,  0x1000,
ota_0,    0,    ota_0,   0x10000,  0x160000,
ota_1,    0,    ota_1,  0x170000,  0x160000,
uf2,      app,  factory,0x2d0000,  256K,
user_fs,  data, fat,    0x310000,  960K,
```

hathach, 2020-08-28:
@tannewt I am trying to shrink the uf2 app, will try to get it down 180KB. I think we can cherry-pick or disable some of the libc and/or the log message. The libc alone is 55KB !!!

```
Total sizes:
 DRAM .data size:    7268 bytes
 DRAM .bss  size:   18016 bytes
Used static DRAM:       0 bytes (      0 available, nan% used)
Used static IRAM:       0 bytes (      0 available, nan% used)
Used stat D/IRAM:   63283 bytes ( 297165 available, 17.6% used)
      Flash code:  107387 bytes
    Flash rodata:   15484 bytes
Total image size:~ 186154 bytes (.bin may be padded larger)
Per-file contributions to ELF file:
             Object File DRAM .data & .bss & other   IRAM   D/IRAM Flash code & rodata   Total
        lib_a-vfprintf.o          0      0       0      0        0      13685      752   14437
       lib_a-svfprintf.o          0      0       0      0        0      13290      750   14040
          msc_device.c.o          0   4160       0      0     4160       4266      152   12738
      lib_a-svfiprintf.o          0      0       0      0        0       9619     1206   10825
       lib_a-vfiprintf.o          0      0       0      0        0       9933      734   10667
                main.c.o          0   4440       0      0     4440        145       38    9063
           flash_hal.c.o          4   4100       0      0     4104        414      420    9042
                usbd.c.o         92    248       0      0      340       5388      224    6292
                 rmt.c.o         36     16       0      0     1642       3549      949    6192
       panic_handler.c.o       2198      0       0      0     3930          0        0    6128
               tasks.c.o         12    660       0      0     4865          0      456    5993
         dcd_esp32s2.c.o          0    180       0      0      180       4813       40    5213
                 msc.c.o          1   2072       0      0     2073        335       20    4501
            lib_a-dtoa.o          0      0       0      0        0       3524       13    3537
             portasm.S.o       1544      0       0      0     1913          0        0    3457
             ringbuf.c.o          0      0       0      0     2934          0      411    3345
            vfs_uart.c.o         80      8       0      0       88       2571      152    2899
          hid_device.c.o          0    140       0      0      140       2356       48    2684
               queue.c.o          0      0       0      0     2057        133      399    2589
           lib_a-mprec.o          0      0       0      0        0       2144      296    2440
          intr_alloc.c.o          8     13       0      0      473       1429      512    2435
                uart.c.o         40      8       0      0       48       1855      477    2428
    esp_image_format.c.o          0      0       0      0        0       1491      884    2375
       vendor_device.c.o          0    468       0      0      468       1332        0    2268
             rtc_clk.c.o        234      0       0      0     1977          0        0    2211
spi_flash_chip_generic.c        238      0       0      0     1805          0        0    2043
       esp_flash_api.c.o         16      0       0      0     1719         52      237    2024
                 vfs.c.o        192     40       0      0      232       1549        5    2018
          flash_mmap.c.o          0    392       0      0     1477         94        0    1963
            ghostfat.c.o        384      4       0      0      388        722      305    1803
  spi_flash_hal_iram.c.o         24      0       0      0     1625          0        0    1649
           tusb_fifo.c.o          0      0       0      0        0       1556        0    1556
 spi_flash_hal_gpspi.c.o         24      0       0      0     1520          0        0    1544
            rtc_init.c.o          0      0       0      0        0       1318       56    1374
                gpio.c.o         32      0       0      0       32       1014      281    1359
        wdt_hal_iram.c.o          0      0       0      0     1351          0        0    1351
            task_wdt.c.o         53     12       0      0       65        913      305    1348
                 sha.c.o          0      0       0      0        0       1024      301    1325
              timers.c.o          8     56       0      0      986          0      212    1262
             memprot.c.o          0      0       0      0      156        967       51    1174
      xtensa_vectors.S.o          0      0       0      0     1173          0        0    1173
led_strip_rmt_ws2812.c.o          0     16       0      0      181        597      377    1171
          multi_heap.c.o          0      0       0      0     1135          0        0    1135
        usbd_control.c.o          0     84       0      0       84        958        0    1126
     xtensa_intr_asm.S.o        512      0       0      0      563          0        0    1075
         periph_ctrl.c.o          8     34       0      0       42        661      272    1017
               panic.c.o        153      5       0      0      857          0        0    1015
          lib_a-fseeko.o          0      0       0      0        0        910        0     910
  memspi_host_driver.c.o        187      0       0      0      696          0        0     883
                 log.c.o          8    260       0      0      302        308        0     878
[... cut ...]
            mac_addr.c.o          0      0       0      0        0         24        0      24
        esp_rom_uart.c.o          0      0       0      0       24          0        0      24
bootloader_flash_config_          0      0       0      0        0         17        0      17
             pthread.c.o          0      0       0      0        0         12        0      12
           lib_a-errno.o          0      0       0      0        0         10        0      10
                crtend.o          0      0       0      0        0          0        8       8
```

tannewt, 2020-08-28:
@hathach I wouldn't worry about it. I think we should leave it at 256k to give us flexibility in the long term.

hathach, 2020-08-29:
@tannewt thanks, indeed, I plan to add webusb/hid aka hf2 support later on as well. I am able to trim down the flash size by 50KB by changing several option #6, still very handy. the static lib for ESP is huge, just blinking example also takes 150KB, I guess it is hard to get any smaller :).

@me-no-dev could I ask how often the Arduino repo make use of nvs partition, if yes and alot, maybe we should also have an nvs as well to make sure all the code run as it is with little modification.

me-no-dev, 2020-08-30:
@hathach NVS is used by many things not only in Arduino, but in IDF as well. Things like WiFi settings and others are stored there by the system. Arduino uses it for it's Preferences class and for EEPROM. You should always have NVS defined :)

hathach, 2020-08-31:
thank you @me-no-dev for nvs explanation. @tannewt maybe we should still keep the NVS, we can reduce the size of it if needed. That will help to run Arduino sketches and use existing "components" from IDF easier.

tannewt, 2020-08-31:
Good idea! What limitations are there on NVS? Can we fit it in the space remaining between `phy_init` and `ota_0`?

I don't expect CircuitPython to use it for wifi credentials because we have the filesystem with source code.

me-no-dev, 2020-09-01:
You can have it before otadata at address 0x9000 with size 0x4000 :) that is the default size and offset that we use

tannewt, 2020-09-01:
Perfect! Will update it.

tannewt, 2020-09-01:
@me-no-dev Do you think we should drop `phy_init`? I don't see it in any Arduino options.

me-no-dev, 2020-09-02:
phy_init is not necessary :)

hathach, 2020-09-07:
@tannewt let's me know when you have the final partition version, I will update the one in uf2 here :).

tannewt, 2020-09-09:
@hathach Here is what I think is final:
```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
# bootloader.bin,,          0x1000, 32K
# partition table,,         0x8000, 4K
nvs,      data, nvs,      0x9000,  20K,
otadata,  data, ota,      0xe000,  8K,
ota_0,    0,    ota_0,   0x10000,  1408K,
ota_1,    0,    ota_1,  0x170000,  1408K,
uf2,      app,  factory,0x2d0000,  256K,
user_fs,  data, fat,    0x310000,  960K,
```

me-no-dev, 2020-09-09:
`nvs,      data, nvs,      0x9000,  20K,` <<< should be 2K, not 20 :)

me-no-dev, 2020-09-09:
Actually I take that back. You need to make it 3K or more (4K is best here). 2K NVS is ReadOnly.

tannewt, 2020-09-09:
Any reason not to fill the first 64k with NVS?

me-no-dev, 2020-09-10:
really, you do not need that much. You are limited to how big one item inside can be (a bit under 2K). But other than that disregard previous comment. my math went wrong. It is 20K difference.

tannewt, 2020-09-25:
This is done in #10

tannewt, 2020-12-01:
I think we should consider moving the factory partition before the ota partitions. This will drastically shrink the size of the combined bin and not overwrite the ota sections then. I think we'll want to overwrite otadata though. Otherwise it'll need to go in it's own 64k block. Unless we can make the factory section a bit less thank full 64k blocks.

hathach, 2020-12-04:
it will make it harder to get it running with ESP32 Arduino, since the app address is fixed at 0x1000 unless we could somehow persuade espressif to do so, which I think it is hard to do. 
https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/platform.txt#L140

@tannewt why don't we flash it using multiple bin with address instead of combining the bin file ?

me-no-dev, 2020-12-04:
What is your partition layout and what will Arduino flash? the OTA?
If the app address is made into an option that defaults to 0x10000 and you will be flashing to the ota slots from Arduino, then all should be fine.

tannewt, 2020-12-04:
> @tannewt why don't we flash it using multiple bin with address instead of combining the bin file ?

It's easier for folks to download and flash a single file. @makermelissa was having trouble flashing one big file over webserial.

hathach, 2020-12-07:
Since OTA0 must be at 0x10000, I think we could only move the tinyuf2 to the region between 0TA0 & OTA1
e.g
```
0x10000 OTA0
0xAAAA  tinyuf2
0xBBBB  OTA1
```
hmm, it should not be a problem with 0xff as padded since that wouldn't trigger a erase/flash operation ? Though I am not sure about webserial, maybe the browser place some litmit of its own.

tannewt, 2020-12-07:
@hathach Let's not bother with this. It's better to match arduino and newer boards will all have UF2 already installed. Thanks!

closed 2020-12-07 by tannewt