adafruit/tinyuf2 #12: Hard Lock-up on Saola-Wroom

julianrendell, 2020-09-27:
Up front: this is probably user error...

I've built and flashed the latest from this repo's master branch.

I am using the latest idf-tools. 

I've enabled level 2 TinyUSB logging and added some extra logging:

```
--- idf_monitor on /dev/cu.usbserial-14430 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x1aa8
load:0x4004c000,len:0x1cd4
load:0x40050000,len:0x1c08
entry 0x4004c454
I (46) boot: ESP-IDF v4.3-dev-1197-g8bc19ba89 2nd stage bootloader
I (46) boot: compile time 19:13:47
I (46) boot: chip revision: 0
I (49) boot.esp32s2: SPI Speed      : 80MHz
I (54) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 4MB
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (87) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (95) boot:  2 ota_0            OTA app          00 10 00010000 00160000
I (102) boot:  3 ota_1            OTA app          00 11 00170000 00160000
I (110) boot:  4 uf2              factory app      00 00 002d0000 00040000
I (117) boot:  5 user_fs          Unknown data     01 81 00310000 000f0000
I (125) boot: End of partition table
I (129) boot: Defaulting to factory image
I (134) boot: JULIAN- boot_index: -1
I (138) boot: JULIAN- RESET_REASON: 1
I (142) boot: JULIAN- boot_index-2: -1
I (147) esp_image: segment 0: paddr=0x002d0020 vaddr=0x3f000020 size=0x037f4 ( 14324) map
I (159) esp_image: segment 1: paddr=0x002d381c vaddr=0x3ffbb360 size=0x019c4 (  6596) load
I (166) esp_image: segment 2: paddr=0x002d51e8 vaddr=0x40022000 size=0x00404 (  1028) load
0x40022000: _WindowOverflow4 at /Users/julian/Documents/MakeItZone_Projects/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (174) esp_image: segment 3: paddr=0x002d55f4 vaddr=0x40022404 size=0x08f50 ( 36688) load
0x40022404: _coredump_iram_end at ??:?

I (192) esp_image: segment 4: paddr=0x002de54c vaddr=0x00000000 size=0x01acc (  6860)
I (194) esp_image: segment 5: paddr=0x002e0020 vaddr=0x40080020 size=0x105f8 ( 67064) map
0x40080020: _stext at ??:?

I (219) boot: Loaded app from partition at offset 0x2d0000
I (219) boot: Disabling RNG early entropy source...
I (231) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (231) cpu_start: cpu freq: 160
I (233) cpu_start: Pro cpu up.
I (290) cpu_start: Pro cpu start user code
I (290) cpu_start: Application information:
I (290) cpu_start: Project name:     uf2-esp32s
I (294) cpu_start: App version:      47c7e86-dirty
I (299) cpu_start: Compile time:     Sep 27 2020 01:03:42
I (305) cpu_start: ELF file SHA256:  c62dcfbf23055695...
I (311) cpu_start: ESP-IDF:          v4.3-dev-1197-g8bc19ba89
I (318) heap_init: Initializing. RAM available for dynamic allocation:
I (325) heap_init: At 3FFC1358 len 0003ACA8 (235 KiB): DRAM
I (331) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (338) spi_flash: detected chip: generic
I (342) spi_flash: flash io: dio
I (346) cpu_start: Starting scheduler on PRO CPU.
I (351) uf2: Hello
USBD init
MSC init
HID init
VENDOR init
I (351) uf2: _part_ota0->type = 0
I (361) uf2: _part_ota0->subtype = 0x10
I (361) uf2: _part_ota0->address = 0x10000
I (361) uf2: _part_ota0->size = 0x160000
I (371) uf2: _part_ota0->encrypted = 0
I (371) uf2: flash->chip_id = 0x204016
I (381) uf2: flash->size = 0x400000
I (381) uf2: block_erase = 0x10000
I (391) uf2: JULIAN-hit usb dev task
I (391) uf2: JULIAN-pre tud task loop
```

The last log message is from main/main.c where I added:

```
void usb_device_task(void* param)
{
  (void) param;
  ESP_LOGI(TAG, "JULIAN-hit usb dev task");
  // RTOS forever loop
  while (1)
  {
    ESP_LOGI(TAG, "JULIAN-pre tud task loop");
    tud_task();
    ESP_LOGI(TAG, "JULIAN-post tud task loop");
  }
}
```

`tud_task()` seems to be locking up. Plugging native USB into a Mac or PC -> nothing.

The board has run CircuitPython fine.

I have tried erasing the flash first and then flashing with the uf2 build.

What's the next step to debug this?

julianrendell, 2020-09-29:
Using @tannewt's version of `esp-idf` included with CircuitPython just works.

There's no obvious compilation errors or other issues with the latest `esp-idf`, but it doesn't work; seems to get locked up in `tinyusb`.

tannewt, 2020-09-29:
@julianrendell I'd suggest trying the esp-idf version my branch is based on. I'd expect it to work. Once you do that, you can binary search the idf changes to see what changed. My guess is the process for initing the USB peripheral or its pins.

julianrendell, 2020-09-29:
@tannewt - yip. My 2nd comment confirmed that it works fine with the esp-idf version you patched in CP :-)

I've changed the title to reflect this is something to do with esp-idf, and which version I used.

I'll reach out to @hierophect and see how his merge of the latest esp-idf is going and if he has any capacity to provide some guidance/mentorship ;-)

hathach, 2020-09-30:
it is probably due to the work in progress with usb in IDF, last time I build with IDF master, I work just fine. I will give it a try later on to see if there is a quick fix. For reference, my local IDF commit is https://github.com/espressif/esp-idf/commit/6c17e3a64c02eff3a4f726ce4b7248ce11810833

hathach, 2020-10-22:
should be resolved with #14

closed 2020-10-22 by hathach