adafruit/tinyuf2 #18: Can't build for my FeatherS2

UnexpectedMaker, 2020-11-03:
Did a brand new fork and clone - got all submodules.
Git status shows I'm all updated and clean.

This is what I get when I try to build for my featherS2...

```
 ❯ make BOARD=unexpectedmaker_feathers2 all                                                                                                                           [16:43:46]
idf.py -B_build/build-unexpectedmaker_feathers2 -DBOARD=unexpectedmaker_feathers2 build
Executing action: all (aliases: build)
Running cmake in directory /Users/seon/tinyuf2/ports/esp32s2/_build/build-unexpectedmaker_feathers2
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DBOARD=unexpectedmaker_feathers2 -DIDF_TARGET=esp32s2 -DCCACHE_ENABLE=0 /Users/seon/tinyuf2/ports/esp32s2"...
-- Found Git: /usr/bin/git (found version "2.24.3 (Apple Git-128)") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/seon/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /Users/seon/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /Users/seon/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Users/seon/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++
-- Check for working CXX compiler: /Users/seon/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s2
Loading defaults file /Users/seon/tinyuf2/ports/esp32s2/sdkconfig.defaults...
Loading defaults file /Users/seon/tinyuf2/ports/esp32s2/boards/unexpectedmaker_feathers2/sdkconfig...
Traceback (most recent call last):
  File "/Users/seon/circuitpython/ports/esp32s2/esp-idf/tools/kconfig_new/confgen.py", line 641, in <module>
    main()
  File "/Users/seon/circuitpython/ports/esp32s2/esp-idf/tools/kconfig_new/confgen.py", line 328, in main
    update_if_changed(temp_file, filename)
  File "/Users/seon/circuitpython/ports/esp32s2/esp-idf/tools/kconfig_new/confgen.py", line 618, in update_if_changed
    with open(destination, "w") as f:
IOError: [Errno 2] No such file or directory: '/Users/seon/tinyuf2/ports/esp32s2/build/sdkconfig'
CMake Error at /Users/seon/circuitpython/ports/esp32s2/esp-idf/tools/cmake/kconfig.cmake:226 (message):
  Failed to run confgen.py
  (/Users/seon/.espressif/python_env/idf4.3_py2.7_env/bin/python;/Users/seon/circuitpython/ports/esp32s2/esp-idf/tools/kconfig_new/confgen.py;--kconfig;/Users/seon/circuitpython/ports/esp32s2/esp-idf/Kconfig;--sdkconfig-rename;/Users/seon/circuitpython/ports/esp32s2/esp-idf/sdkconfig.rename;--config;/Users/seon/tinyuf2/ports/esp32s2/build/sdkconfig;--defaults;/Users/seon/tinyuf2/ports/esp32s2/sdkconfig.defaults;--defaults;/Users/seon/tinyuf2/ports/esp32s2/boards/unexpectedmaker_feathers2/sdkconfig;--env-file;/Users/seon/tinyuf2/ports/esp32s2/_build/build-unexpectedmaker_feathers2/config.env).
  Error 1
Call Stack (most recent call first):
  /Users/seon/circuitpython/ports/esp32s2/esp-idf/tools/cmake/build.cmake:442 (__kconfig_generate_config)
  /Users/seon/circuitpython/ports/esp32s2/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:35 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/seon/tinyuf2/ports/esp32s2/_build/build-unexpectedmaker_feathers2/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
make: *** [all] Error 2
```

I've tried with python2 and python 3, and this is using the export.sh from CircuitPython latest.

UnexpectedMaker, 2020-11-03:
damn, gotta use idf.py - sorry!

closed 2020-11-03 by UnexpectedMaker

hathach, 2020-11-03:
actually it is meant to be able to build with `make BOARD=` as well, it just have an issue with build directory. #20 should fix this. Thank you for raising the issue.

UnexpectedMaker, 2020-11-03:
Oh, ok, yeah I used to make it with `make BOARD=`, but I saw the instructions in the esp32s2 readme and thought that was not supported anymore. Glad to hear you've fixed it anyway!

hathach, 2020-11-03:
docs is a bit old, I always be later with doc :(

UnexpectedMaker, 2020-11-03:
> docs is a bit old, I always be later with doc :(

Haha, I'm the same. Always leave docs etc to last - or never ;)

PR #20 (2020-11-03): fix esp32s2 build/sdkconfig
fix #18