adafruit/tinyuf2 #21: Wrong partition sizing for FeatherS2

UnexpectedMaker, 2020-11-04:
Using the new merged UF2 bootloader, then flashing with CPY 6 rc.1, We are only seeing 960KB free on the drive instead of the 12(ish)MB we should see. Using the old UF2 bootloader (pre tinyUF2 merge) it works correctly.

The FeatherS2 sdkconfig is definitely pointing to the 16MB partition scheme, but it's it doesn't seem to be working.

Have had more than 4 people confirm this so far, and can repro this myself on several boards.

UnexpectedMaker, 2020-11-04:
For clarification, here are volume stats from 2 boards... 1 running new UF2 built from latest source pulled 5 mins ago, and one from previous UF2 bootloader pre merge. Both with rc.1 installed.

[image]
[image]

hathach, 2020-11-05:
> For clarification, here are volume stats from 2 boards... 1 running new UF2 built from latest source pulled 5 mins ago, and one from previous UF2 bootloader pre merge. Both with rc.1 installed.

Time is rather vague, 
1. can you confirm you are latest with the commit hash. 
2. Please try again with fullclean, remove buld folder then try again with `make BOARD=`, also check the partition table to see if it is generated correctly here. I think the bootloader can only dictate this far.
3. Also make sure you use the same circuitpython uf2 for both bootloader version 

```
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,20K,
otadata,data,ota,0xe000,8K,
ota_0,app,ota_0,0x10000,2M,
ota_1,app,ota_1,0x210000,2M,
uf2,app,factory,0x410000,256K,
user_fs,data,fat,0x450000,11968K,
*******************************************************************************
```

UnexpectedMaker, 2020-11-08:
Just realised I'd not replied to this. I couldn't get my local repo to build this properly, so I killed it off and started again, and it worked. So I *somehow* got my local folder in a tizz.

It's all working fine now :)

closed 2020-11-08 by UnexpectedMaker

hathach, 2020-11-08:
no problem at all, we are always busy with something else :) . It is probably not-clean object that is put in default build folder from other board.