adafruit/tinyuf2 #26: Hardware Status

nxf58843, 2020-11-13:
What is the status of various hardware ports, specifically mimxrt10xx?
Wondering how stable it is before adding additional ports.

hathach, 2020-11-14:
it is working, the project is still in early, there will be more functionality and board API changes etc. But the basic function of upgrading with uf2 works for all ports. Which port you are looking to add, is it a new port or just a new board ?

nxf58843, 2020-11-14:
I want to add LPC55.  You already have it in TinyUSB.  
I have UF2 working with the NXP SDK, but I want to migrate to TinyUSB.  
I generated a magic number for the family:  NXP LPC55xx - 0x2abc77ec (waiting for pull request from UF2)

hathach, 2020-11-15:
yeah, that is great to hear, feel free too submit an pr for lpc5500 series. I haven't written the porting guide at the moment. But it basically the `board_api.h` . I have an xpresso 55s69 and could test it out myself. You don't have to wait until everything is complete to submit PR, just create a draft one so that we could further discuss how it should implemented, should you need my feedback.

closed 2020-11-19 by hathach