adafruit/tinyuf2 #27: building error

aleggon, 2020-11-18:
**I followed every step of the tutorial, but I always get this error when I try to compile the boot loader for the esp32-s2:**

_[29/72] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log_noos.c.obj
FAILED: esp-idf/log/CMakeFiles/__idf_log.dir/log_noos.c.obj 
/home/alexandre/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc  -Iconfig -I/home/alexandre/esp/esp-idf/components/log/include -I/home/alexandre/esp/esp-idf/components/esp_rom/include -I/home/alexandre/esp/esp-idf/components/esp_common/include -I/home/alexandre/esp/esp-idf/components/xtensa/include -I/home/alexandre/esp/esp-idf/components/xtensa/esp32s2/include -I/home/alexandre/esp/esp-idf/components/esp32s2/include -I/home/alexandre/esp/esp-idf/components/soc/include -I/home/alexandre/esp/esp-idf/components/soc/esp32s2/. -I/home/alexandre/esp/esp-idf/components/soc/esp32s2/include -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -fno-stack-protector -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3-dev-1720-g494a124d9\" -DESP_PLATFORM -DBOOTLOADER_BUILD=1 -MD -MT esp-idf/log/CMakeFiles/__idf_log.dir/log_noos.c.obj -MF esp-idf/log/CMakeFiles/__idf_log.dir/log_noos.c.obj.d -o esp-idf/log/CMakeFiles/__idf_log.dir/log_noos.c.obj -c /home/alexandre/esp/esp-idf/components/log/log_noos.c
/home/alexandre/esp/esp-idf/components/log/log_noos.c:17:10: fatal error: soc/cpu.h: No such file or directory
 #include "soc/cpu.h"  // for esp_cpu_get_ccount()
          ^~~~~~~~~~~
compilation terminated.
[30/72] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log.c.obj
[31/72] Building C object esp-idf/log/CMakeFiles/__idf_log.dir/log_buffers.c.obj
ninja: build stopped: subcommand failed.
[160/1042] Building CXX object esp-idf/nvs_flash/CMakeFiles/__idf_nvs_flash.dir/src/nvs_partition_manager.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1_

**I do not have knowledge enough on the subject to discover what I am doing wrong.

Can anybody help me?**

hathach, 2020-11-18:
you need to use the included submodule of idf, just install and and then export for that version. IDF changes often moving file that break the compilation and the tinyuf2 actually doesn't use much of the IDF modules (mostly basic driver).   
https://github.com/espressif/esp-idf/tree/8bc19ba893e5544d571a753d82b44a84799b94b1 

I will put up some doc for this later on.

skieast, 2020-11-18:
Here is a gist with the full build for an esp32s2 target.
It is using the fish shell commands but the equivalent bash commands are similar
https://gist.github.com/skieast/9184e9ecae6a1245114292d6b16b5fe6

Also documented here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/#get-started-set-up-env

aleggon, 2020-11-18:
It worked. Thank you so much!

closed 2020-11-18 by aleggon