adafruit/tinyuf2 #30: Tinyuf2 bootloader size

Pillar1989, 2020-11-24:
I know each board is different, can you give us some reference sizes for  typical board?

hathach, 2020-11-24:
As you said, there is really no general size for all. Though you could look at the ci build log of a specific board for its size (at the end of build step) https://github.com/adafruit/tinyuf2/actions

Pillar1989, 2020-11-25:
[image]
Minimum 49K, this is not tIny at all.

hathach, 2020-11-25:
Which board you are talking about, you should check its build log in ci as I mentioned earlier rather than looking at the artifacts which bundles several binaries. Also feel free to make your own bootloader.

closed 2020-11-25 by hathach

Pillar1989, 2020-11-25:
@hathach  I'm talking about the stm32f411ve_discovery board, I downloaded artifacts and its bin file, which is only 13.2kB. This is great, hopefully in the future, I can keep it under 8kB.  We plan to port it to this chip.
https://www.st.com/resource/en/datasheet/stm32h735ag.pdf