adafruit/tinyuf2 #39: Start generating releases

dhalbert, 2020-12-16:
Could we start producing numbered releases?

My motivation is that we have some version of TinyUF2 on this Learn Guide page: https://learn.adafruit.com/adafruit-magtag/install-uf2-bootloader, but I'm not sure it included the SCSI completion fix. Some MagTag Windows users are seeing the BOOT drive disappearing too quickly during UF2 copies, and I think that fix might help.

hathach, 2020-12-16:
Yeah sure, I am off for vacation until next week. I will do as soon as I get back.

hathach, 2020-12-23:
bootloader is version and assets are uploaded to release https://github.com/adafruit/tinyuf2/releases/tag/0.1.0

```
TinyUF2 Bootloader 0.1.0 - esp-idf (v4.3-dev-1197-g8bc19ba89) tinyusb (0.7.0-45-g4a2baf40)
Model: Espressif Saola 1R WROVER
Board-ID: ESP32S2-Saola1R-v1.2
```

closed 2020-12-23 by hathach