adafruit/tinyuf2 #42: Porting guide

nxf58843, 2020-12-21:
The ports structure here is different than the structure in tinyUSB.  Is there a straightforward way to bring an example from tinyUSB into tinyUF2 to add support?  I would like to add LPC55 which is already supported in tinyUSB.  I know that I will need to add the flash routines, I'm mostly looking for recommendations to keep the structure maintainable.
The LPC55 has internal flash, so it is quite different from the i.MX RT parts.

hathach, 2020-12-22:
yeah, the structure is somewhat similar but different. The tinyusb is more generic and this repo as application is much more specific to make adding board within a port much easier by a simple `board.h` file. I could help by adding `template` port so that you could start to get it running with lpc55.

gsteiert, 2020-12-22:
Yes, it would be helpful to have a template showing how you would go about it to point me in the right direction.  Thanks

charkster, 2020-12-22:
I would like to use the template too. My stm32f411ce_blackpill board works great with the tinyusb examples, but I have not been successful getting tinyuf2 working. I have made small changes to the included stm32f411ve_discovery board as it is almost identical to my blackpill board, but no luck.
Another difference is that tinyusb uses a *.ld linker script and I can't see one here in tinyuf2.

hathach, 2020-12-23:
@charkster the template is mostly for new port, for stm32f411ce, you should be able to copy-then-modify from the stm32f411disco board. There is need to read-then-modify the makefile if needed to get it work since the current development stage is still pretty early.

charkster, 2020-12-24:
> @charkster the template is mostly for new port, for stm32f411ce, you should be able to copy-then-modify from the stm32f411disco board. There is need to read-then-modify the makefile if needed to get it work since the current development stage is still pretty early.

Thanks. I am not sure if the stm32f411ve discovery board even works. The board files are definitely modified versions of the feather_stm32f405_express ones. I would expect that the feather works as it is an official Adafruit product, but the discovery board... who knows.

hathach, 2020-12-24:
I have just added the `template_port` #44, though it is very basic. Therefore I put a bit more effort to add initial work for lpc55 port here https://github.com/adafruit/tinyuf2/tree/add-lpc55-port/ports/lpc55 . It isn't compiled just yet, still lacking double tap linker symbol etc, but soon would be buildable.

gsteiert, 2020-12-24:
I've been working on this as well.  I think we will need to update the NXP library for the lpc55.  There are new functions in fsl_iap that are needed.  I'll try to merge my work into your template. Thanks.

hathach, 2020-12-25:
#45 add initial ground work for lpc55 port with 55s69 and 55s28 both can enumerate as uf2 drive. Most of other functionality such as indicator (PWM LED, RGB), double tap, flash aren't implemented yet. Hopefully this could help you out with the port.

PR #45 (2020-12-25): Add lpc55 port
This is ground work for lpc55 port, LPCXpresso 55s69 and 55s28 board enumerated as msc. PWD LED, RGB, double tap, flash api, self-update are not implemented yet.

close #42
closed 2020-12-25 by hathach