adafruit/tinyuf2 #59: Migrate NXP library to official git repo

nxf58843, 2021-01-19:
Now that NXP has an official public github release of their SDK, we should migrate the NXP platforms to use it.
https://github.com/NXPmicro/mcux-sdk

hathach, 2021-01-20:
great news, thank you for opening the issue. I will try to see if we could quickly migrate to nxp repo.

gsteiert, 2021-01-20:
I was able to migrate my examples (which were based on tinyusb)
Many of the drivers have their own directories so the include paths and driver paths need to be updated.
https://github.com/gsteiert/double_m33_examples
All the build changes were in board/board.mk
One other thing, there are many parts supported by this package so it is fairly large.

hathach, 2021-01-20:
great to hear that, if it is just matter of changing path, I guess we could do it fairly quickly. Though since it does not change the functionality (hopefully) we could migrate whenever we have time :)

PR #115 (2021-04-29): Use nxp official sdk
fix #59 use official NXP MCU SDK as submodules. @nxf58843
closed 2021-04-29 by hathach