adafruit/tinyuf2 #61: Release self update uf2 files separately

Neradoc, 2021-01-23:
On circuitpython.org the "UF2 bootloader" section for SAMD ports links to the self update uf2 directly, with links like *https://github.com/adafruit/uf2-samdx1/releases/download/v3.10.0/update-bootloader-circuitplay_m0-v3.10.0.uf2*

In order to add the Bootloader section for the ESP32S2 ports, I believe it would be better if the corresponding self-update uf2 files is available separately from the zip in the release, in a similar fashion as above.

hathach, 2021-01-24:
I don’t see any reasons to have them upload separately. I like 1 zip file per board as it is. What is your reason/scenario that requires it to be uploaded separately ?

Neradoc, 2021-01-24:
So people on circuitpython.org can get the UF2 on it's own, like the other UF2 files on the site.
I totally agree it's not necessary, it was more a matter of consistency with the atsamd bootloader in my mind.

hathach, 2021-01-24:
People won’t update bootloader often enough. I think we are good for now, would come back later if needed. Thank you for opening the issue.

hathach, 2021-01-24:
Hmm, I take it back. It may be needed. @dhalbert what do you thonk

dhalbert, 2021-01-24:
in the `uf2-samdx1` releases, the files are all separated, so it was easy for me to link to them directly. We can either separate the file or change the documentation paragraph in circuitpython.org to say that you need to look inside the .zip for an `update-*` file. Right now it says this:

[image]

We have had cases where we wanted people to update their bootloaders to prevent bricking the boards, which is very difficult for most people to recover from. So we wanted to make it as easy as possible.

hathach, 2021-01-25:
got it, I will have  `update-*.uf2` uploaded separated in release/build asset to make everything easier for everyone.

hathach, 2021-02-01:
Since `action/upload-artifact` does always **zip** the file, `update-uf2` is only uploaded to release which allow to upload raw binary.  
https://github.com/adafruit/tinyuf2/commit/dcbd2841ea26f1d4c7323b68bbd8d8284d235a4a#diff-5c3fa597431eda03ac3339ae6bf7f05e1a50d6fc7333679ec38e21b337cb6721R158

hathach, 2021-02-01:
it will be available in the next release, thank you for raising the issue.

closed 2021-02-01 by hathach

hathach, 2021-02-02:
update: I have release 0.2.1 for this issue https://github.com/adafruit/tinyuf2/releases/tag/0.2.1

PR #65 (2021-02-01): Release update uf2 seperately
fix #61 
- `update-tinyuf2` is uploaded separately in release assets. 
- The zip file still contains the `update-uf2` as it is before.
- build artifact does not include update-uf2 since it is zipped anyway.