adafruit/tinyuf2 #63: RT1060 EVK issues

nxf58843, 2021-01-28:
I'm having a couple issues with the imxrt1060 port.
First, double tap is not activating the bootloader.
Second, the fcfb origin for RT1060 does not seem to be correct.  The linker scripts in the NXP SDK place it at 0x0
If you load a valid program (like hello world from the SDK) first, you will not see the issue because the tinyuf2 image gets written after the valid FCFB, but if you mass erase the part and load the tinyUF2 image, when the part powers on it will fail to the ISP ROM.

hathach, 2021-01-28:
are you testing with imxrt 1060 evk ? last time I checked, it stills works. I am currently busy with rp2040 work, will pull out the board to test with when possible.

nxf58843, 2021-01-29:
I am testing with the RT1060 EVK.  I have uncovered the source of some of the issues.  The POR button actually cuts power and corrupts the RAM.  The target reset button SW9 by the debugger will not corrupt the memory, but it only works when the debugger is plugged in, unless you move J43 to pins 2-3.  

Move J43 to pins 2-3 and double tap works on SW9 (with or without debugger plugged in). 

I'm looking at trying to use the SNVS-GPR as a more reliable method, but it is a work in progress.  I'm also looking at resolving the FCFB address issue.

hathach, 2021-01-29:
that is great to hear, thank you for your update.

nxf58843, 2021-03-18:
This was fixed in PR #76

closed 2021-03-18 by nxf58843

ladyada, 2021-03-18:
thanks for closing :)