adafruit/tinyuf2 #74: cannot access I2C after soft reboot with UF2 bootloader

jerryneedell, 2021-02-11:
When the UF2 bootloader is installed on my metro-esp32s2 -- I am unable to access an I2C device after a software reboot
I can access after a hard reset but then it fails after a soft reboot if I try to access it again.

see issue at https://github.com/adafruit/circuitpython/issues/4147

The issue does not occur if I do not use the UF2 Bootloader on the Metro-esp32s2

ladyada, 2021-02-11:
do ya have a logic analyzer trace of the scl/sda lines? it may be illuminating

jerryneedell, 2021-02-11:
Yes -- I'll see what I can find.

jerryneedell, 2021-02-11:
With UF2 bootloader -- this is a "normal" I2C accessing the SHTC3 sensor
[image]

This is the failed aces after a soft reboot -- no activity on SDA

[image]

ladyada, 2021-02-11:
what the...?

ladyada, 2021-02-11:
wanna see if you can query the i2c SDA pin mux? thats the only thing i can imagine it is

jerryneedell, 2021-02-11:
Can you give some instructions. I don't understand what you are asking for. sorry...

ladyada, 2021-02-11:
lets wait till @hathach is back from TET holiday - could you verify that the other 'weird i2c issues' with esp32s2 in circuitpython are resolved when uf2 is not used?

jerryneedell, 2021-02-11:
I'll take a look.

jerryneedell, 2021-02-11:
The issue with the  mcp9600 https://github.com/adafruit/circuitpython/issues/3894
occurs on my metro-esp32s2 even without the UF2 bootloader so does not appear to be related.
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Adafruit Metro ESP32S2 with ESP32S2
>>> import mcp9600_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mcp9600_simpletest.py", line 12, in <module>
  File "adafruit_mcp9600.py", line 178, in __init__
  File "adafruit_mcp9600.py", line 178, in __init__
OSError: [Errno 19] Unsupported operation
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Adafruit Metro ESP32S2 with ESP32S2
>>> import mcp9600_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mcp9600_simpletest.py", line 12, in <module>
  File "adafruit_mcp9600.py", line 165, in __init__
ValueError: No I2C device at address: 67
>>> 

```

jerryneedell, 2021-02-11:
the test code in https://github.com/adafruit/circuitpython/issues/4046  also causes a crash - drop USB and resets
```
import board
import busio
print(dir(board))
i2c = busio.I2C(board.SCL, board.SDA)

def printdevices():
    i2c.try_lock()
    # Find the I2C devices available.
    devices = i2c.scan()
    print("I2C devices:", len(devices))
    for device in devices:
        print(hex(device))
    i2c.unlock()
    print("I2C unlocked! ")

printdevices()
```

ladyada, 2021-02-11:
mcp9600 is a bad chip to use for testing, use something that is 'normal' acting

jerryneedell, 2021-02-11:
> mcp9600 is a bad chip to use for testing, use something that is 'normal' acting

Agreed! I was just trying to reproduce the issue. So far, not having issues with other boards - shtc3 or bme680

makermelissa, 2021-02-19:
Hello, I think I figured out something in regards to this. I hooked up a project that used a bme280 and pm25 sensor (to a MagTag). It always worked fine when I initialized the pm25 sensor first, but failed as described when I did the bme280 sensor. I noticed in the library that the main difference between the init of the drivers is the pm25 is attempted to be probed multiple times and the bme280 is attempted to only be probed once. I made a local change in my bme280 lib to attempt multiple times as well and it consistently worked.

So I hooked up a logic analyzer and it appears to consistently need 2 probes to start communicating. Here are the logic analyzer screenshots.
At a high level, here's what it looks like:
[image]

The first blip looks like this. I think it's just i2c being set up or something.
[image]

Next it does 2 of these. I think this is the probe because I had a delay in there originally (which turned out to be unnecessary) and that altered the time between these packets.
[image]

After that, the chip responds just fine.

makermelissa, 2021-02-19:
This may be more circuitpython related though.

hathach, 2022-07-11:
I guess this is fixed now in circuitpython and safe to close, right ?

makermelissa, 2022-07-11:
I think it should probably be re-tested first. @jerryneedell did you want to try re-testing? I don't have the sensors with me at the moment.

hathach, 2023-12-15:
quick check, could we close this now ? @makermelissa @jerryneedell

makermelissa, 2023-12-15:
I'm pretty sure this isn't an issue anymore. We can always re-open if it is.

closed 2023-12-15 by makermelissa

jerryneedell, 2023-12-15:
Thanks! For some reason, I never saw the earlier prompts to me. Sorry. I was not ignoring you!

hathach, 2023-12-15:
No problem at all. I forgot about this as well :)