adafruit/tinyuf2 #75: stm32f411ce blackpill appears to not program new uf2 properly

charkster, 2021-02-15:
I have created a new stm32f411ce_blackpill board using the stm32f401ce_blackpill board files as a template. I have updated the board.mk and board.h to use the proper device name, and after flashing the board it boots properly (usb hid device). When I drag-n-drop the new uf2 file to flash it to the board I am not getting the expected result.

I don't know if this is related but... I have found that in board.mk I need to keep `-DHSE_VALUE=25000000U` or the new uf2 bootloader will not appear as a hid device. This is not expected as the stm32f411ve_discovery board uses `-DHSE_VALUE=8000000U` in board.mk.
I have also updated my new board.h file to specify `#define BOARD_FLASH_SIZE  (512 * 1024)` and `#define BOARD_FLASH_SECTORS 8` (which is different to the stm32f401 board, as it is 256k flash with 6 sectors).

I am using the st-flash utility to write the bin file to 0x8000000 on my stm32f411ce_blackpill... that appears to be working as expected.

Should the bootloader be writing the drag-n-dropped uf2 file to 0x8010000? I am assuming that when 0x8010000 has a valid value then the uf2 bootloader will start execution there (and not appear as a hid device).

Another question, how would I get back to uf2 bootloader mode again? When I release the BOOT0 button after the RST_N button has been released, I will enter into DFU mode (I assume this is ROM execution). Is there a button sequence to ensure that I enter UF2 bootloader mode (like pressing the BOOT0 button or the "KEY" button)?
Thanks.

charkster, 2021-02-15:
This is from dmesg when I see the hid device.

```
[ 6459.105034] usb 1-1.4: new full-speed USB device number 10 using xhci_hcd
[ 6459.254544] usb 1-1.4: New USB device found, idVendor=239a, idProduct=005d, bcdDevice= 1.00
[ 6459.254558] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6459.254569] usb 1-1.4: Product: STM32F411CxUx
[ 6459.254580] usb 1-1.4: Manufacturer: STM32
[ 6459.254590] usb 1-1.4: SerialNumber: 2B002C000351303033343331
[ 6459.260507] usb-storage 1-1.4:1.0: USB Mass Storage device detected
[ 6459.261174] scsi host0: usb-storage 1-1.4:1.0
[ 6459.269817] hid-generic 0003:239A:005D.0004: hiddev96,hidraw0: USB HID v1.11 Device [STM32 STM32F411CxUx] on usb-0000:01:00.0-1.4/input1
[ 6460.317917] scsi host0: scsi scan: INQUIRY result too short (5), using 36
[ 6460.317936] scsi 0:0:0:0: Direct-Access     Adafruit UF2 Bootloader   1.0  PQ: 0 ANSI: 2
[ 6460.318513] sd 0:0:0:0: Attached scsi generic sg0 type 0
[ 6460.319371] sd 0:0:0:0: [sda] 65801 512-byte logical blocks: (33.7 MB/32.1 MiB)
[ 6460.319718] sd 0:0:0:0: [sda] Write Protect is off
[ 6460.319735] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[ 6460.325414] sd 0:0:0:0: [sda] No Caching mode page found
[ 6460.325428] sd 0:0:0:0: [sda] Assuming drive cache: write through
[ 6460.391805]  sda:
[ 6460.393914] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

charkster, 2021-02-15:
CURRENT.UF2 appears to be 1MB in size on the STM32F411CE_BLACKPILL. On my samd21 the CURRENT.UF2 appears as 512.KiB. This looks proper as the on-chip flash for the STM32F411CE is 512KiB and the samd21 has 256KiB (the CURRENT.UF2 appears as twice the available flash memory size).

charkster, 2021-02-15:
In looking at the stm32f411ce datasheet, there are 7 flash sectors.

Sector 0 0x0800 0000 - 0x0800 3FFF 16 Kbytes
Sector 1 0x0800 4000 - 0x0800 7FFF 16 Kbytes
Sector 2 0x0800 8000 - 0x0800 BFFF 16 Kbytes
Sector 3 0x0800 C000 - 0x0800 FFFF 16 Kbytes
Sector 4 0x0801 0000 - 0x0801 FFFF 64 Kbytes
Sector 5 0x0802 0000 - 0x0803 FFFF 128 Kbytes
Sector 6 0x0804 0000 - 0x0805 FFFF 128 Kbytes
Sector 7 0x0806 0000 - 0x0807 FFFF 128 Kbytes

I updated the board.h file and re-ran but no improvement.
I will keep `#define BOARD_FLASH_SECTORS 8`

charkster, 2021-02-16:
I am seeing that the uf2 bootloader is written from 0x8000000 to 0x8003880.

[image]

I am seeing that nothing has been written to 0x8010000.
[image]

charkster, 2021-02-17:
`-DHSE_VALUE=25000000U` in board.mk is correct because the board uses a 25MHz oscillator (just like the STM32401 blackpill).
Good to know that refers to the board oscillator.

charkster, 2021-02-17:
I programmed various values into 0x8010000, and then did a drag-n-drop of the new uf2 file to see if the bootloader would at least erase the 0x8010000 sector... it did not get erased. I am certain that this is a valid uf2 file (it was created using the python tool, and that same tool has made successful uf2 files for the samd21). Is there a jlink logger that I can enable to help see where the first problem is occuring?

charkster, 2021-02-18:
Here is the RTT logfile with the USB attach and the uf2 file drag-n-drop.
[tinyuf2_stm32f411ce_blackpill_rtt_logfile.txt](https://github.com/adafruit/tinyuf2/files/5999753/tinyuf2_stm32f411ce_blackpill_rtt_logfile.txt)

charkster, 2021-02-18:
With LOG=1 no errors are reported. With LOG=2 I am definitely seeing the logfile reacting to the new UF2 file drag-n-drop. As no errors are being reported, I am assuming that the flash erase is the issue (as I can have random values at 0x8010000 and they are never erased). If the UF2 file had a problem with size or formatting, I would hope that with LOG=1 something would have been indicated.

hathach, 2021-02-18:
thanks for the issue, I am currently working on other projects. Will try to help whenever I could. Feel free to update any finding meanwhile

charkster, 2021-05-30:
I had a chance to look at this again. I manually programmed data to 0x8010000 so that it was not blank. Ran the RTTViewer with LOG=2 and did a drag-n-drop of the uf2 file. The drag-n-drop now shows a message box that says it is waiting for the write to complete (but hangs). The last RTT log message is "Erase: 08004000 size = 16384".  I will insert some more log messages in "board_flash.c" to see if I can identify the code which causes the hang. I did check the `HAL_FLASH_Unlock()` routine and it appears to be correct... I checked the board.mk file to make sure the correct board was specified, and that the board clock frequency is correct.

hathach, 2021-05-31:
maybe you could try to change the flash latency to see if it is an issue.

charkster, 2021-05-31:
I switched parts and manually erased 0x08004000 sector 1. I verified that it was erased. I then programmed tinyuf2 boot loader using j-link. I noticed that 0x08004000 had data in it. I thought that tinyuf2 bootloader had to fit inside 0x08000000 to 0x08003FFF. I believe I am getting stuck when I am executing from 0x08004000 and trying to erase it at the same time.

charkster, 2021-05-31:
I compiled again without the "LOG=2 LOGGER=rtt" and I fit into 0x08003FFF. So I believe my debugging efforts were causing me to hang. I believe if I want to RTT log, I will need to modify "board_flash.c" to have it not erase sector 1 (like it avoids sector 0).
[image]

charkster, 2021-05-31:
After looking at my original uf2 file that I was testing with, it was created with a base address of 0x08000000, which conflicts with the bootloader... I believe this is why sectors below 0x080100000 were getting erased. I manually run the uf2conv.py script now and specify the 0x08010000 as the base address (as confirmed in the following screen capture). There still is no flash write.
Would it not be better/safer to use the APP_LOAD_ADDRESS defined in board_flash.c and ignore the address in the UF2 file? It appears that is done when reading the app, but not with the flashing/writing of the app.
[image]

charkster, 2021-06-01:
I checked my testing uf2 file again for the expected stm32f4 family_id (as defined in the Makefile) UF2_FAMILY_ID = 0x57755a57... and it was not matching... which explains why no flash writes were happening. I am seeing my expected writes at 0x08010000 now.
Here is my manual uf2 conversion script:
uf2conv.py -o stm32f411blackpill_app.uf2 -b 0x08010000 -f 0x57755a57 stm32f411blackpill_app.bin

Now my issue is that the bootloader is not passing execution to the app at 0x08010000. It is always appearing as a flash drive device.

charkster, 2021-06-01:
After looking at my app's bin file and the values at 0x08010000, they appear identical (hooray!). I was expecting that the tinyuf2 bootloader would look at the first couple of values at APP_LOAD_ADDRESS to see if it was programmed... and then transfer execution.
Without the double press method, I guess it would be impossible to restart the bootloader and skip the transfer of execution to the app (so that the app could be over-written with a new app).
Does new code need to be written before execution can be transferred to the app?
Sorry for all the verbose comments.

hathach, 2021-06-01:
OK, so sum up, the DFU process is complete, all the binary data is written to correct address. However, bootloader just does not perform jump to app. This is probably easier to fix. Btw, I don't see the f411ce blackpill in the board list, could you make PR to add it, so that I could try to see if there is any issue with the configuration in compare with f411 discovery.

hathach, 2021-07-29:
it has been a while, I have made an pr #137 that fixed an issue with updating circuitpython firmware, it could also fix this issue as well. Please give it a try to see if that is indeed the case.

charkster, 2021-07-30:
Thanks for letting me know. I pulled the latest changes and built stm32f411ce_blackpill. I did a chip erase and programmed the new bin file. It booted as a flash drive as expected and I then copied over the uf2 file. I disconnected the USB and connected it again. It booted as a flash drive... so branch execution to the app did not occur. I checked the flash memory in stm32 cube programmer and saw that location 0x08010000 was programmed properly. I am not sure if any STM32 control registers need to be written to before execution is allowed to branch. If execution to the app was performed I would not expect to see the MCU to appear as a flash drive... but the MCU is always appearing as a flash drive.

[image]

hathach, 2021-07-30:
1. @charkster could tell me more about the application you are compiling, please make sure you compile with linker script starting from 0x08010000 (instead of 0x08000000 then shift it with uf2, I am not so sure, it may not be the same). 
2. Also please try with a super simple app such as blinky just for verification first.
3. Could you provide your uf2, I will try to test to see if I could spot something unusual. Note: preferably an simply blinky with correct address as mention in 1 and 2

charkster, 2021-07-30:
@hathach 1. I am using the TINYUSB example (USBTMC device). I had been using the uf2 make option, but it is reporting that it is using 0x08000000. I updated STM32F411CEUx_FLASH.ld to use 0x08010000.
`FLASH (rx)      : ORIGIN = 0x08010000, LENGTH = 512K`
After using this file, it appears that execution was transferred to the app. The app did not work correctly, but it did not appear as a flash drive any more.
I believe I can close this issue, as the bootloader is working. Maybe my apps are relying on POR values in the peripheral registers, and the tinyuf2 bootloader has changed some register values.
Making UF2 files for the SAMD21 and SAMD51 are really easy. I can take a bin file from an arduino build and use the uf2conv.py script. I can even take the tinyusb examples and make working uf2 files.
Thank you for your help.

closed 2021-07-30 by charkster

hathach, 2021-07-30:
thanks for confirmation, depending on the linker, or the code could reference to absolute address somewhere. especially the ISR vector. The app run but look like it hit the hardfault, the tinyusb example uf2 version is general option, I didn't test it with all ports/boards. Glad we finally find the root cause.