adafruit/tinyuf2 #89: Enable capability to use >1 sector per cluster in ghostfat

henrygab, 2021-03-13:
@kaetemi ... maybe I'm missing something, but I think there needs to be a way to enable the use of the new >1 sector per cluster support.

For example, shouldn't the following:

https://github.com/adafruit/tinyuf2/blob/15fa707341df119583e5241e3acf91320859be82/src/uf2.h#L45-L47

Be changed to:

```C
#ifndef CFG_UF2_FLASH_SIZE
    #define CFG_UF2_FLASH_SIZE          (4*1024*1024)    // TODO absolute max flash size
#endif
#ifndef CFG_UF2_NUM_BLOCKS
    #define CFG_UF2_NUM_BLOCKS          (0x10109)        // just under 32MB
#endif
#ifndef CFG_UF2_SECTORS_PER_CLUSTER
    #define CFG_UF2_SECTORS_PER_CLUSTER (1)
#endif
```

If so, I'll generate a quick PR for this...

kaetemi, 2021-03-13:
Yeah, that would be more practical. :)

kaetemi, 2021-03-13:
Can we include the tested large value combination in the comments?

hathach, 2021-03-13:
Yeha, there is still lots of TODO, we will move it to per ports/boards later on. But this look good

hathach, 2021-03-15:
> Can we include the tested large value combination in the comments?

Yeah, I think having a working/tested value combination would be great. It is removed recently (probably mistaken as obsolete value). Would you mind added it, maybe along with a bit of detail on your setup e.g like this combination appear as 1.5 GBs disk and tested on abc hardware would be helpful as well.

kaetemi, 2021-03-15:
```
#define CFG_UF2_FLASH_SIZE          (256*1024*1024)
#define CFG_UF2_NUM_BLOCKS          (0x300000)       // 1.5 GB
#define CFG_UF2_SECTORS_PER_CLUSTER (64)
```

That's for 256MB flash, so 2x a 512MB UF2 file, and plenty for space remaining for the two text files (or for the remainder of a UF2 file with non-flash content).

Tested on RP2040, for flashing BT815-series graphics chip. (Maybe add a note on the memory required for the buffer that keeps the written block flags... `CFG_UF2_FLASH_SIZE / 256 / 8` iirc? That's the bottleneck, it seems.)

hathach, 2021-03-16:
> ```
> #define CFG_UF2_FLASH_SIZE          (256*1024*1024)
> #define CFG_UF2_NUM_BLOCKS          (0x300000)       // 1.5 GB
> #define CFG_UF2_SECTORS_PER_CLUSTER (64)
> ```
>
> That's for 256MB flash, so 2x a [...]

Great, would you make an PR to add these comment along with the note. `CFG_UF2_FLASH_SIZE` is from original uf2 bootloader for nrf52, there is still work to generalize this project. It is still in early stage, so yeah, we will revise it and update/remove it when needed.

henrygab, 2021-03-17:
@kaetemi -  It was my PR that removed the large commented-out values.  Sorry!

@hathach -  Having the CFG_* values is important ... I am close to having self-validation code working, at least to the point I must hand it to you for proper integration.  I will open a separate issue/PR for that. :)

hathach, 2021-03-17:
@henrygab ah ok, actually, I have no plan/time to remove it anytime soon. But rather will evolve slowly when needed :) I am looking forward to your PR.

henrygab, 2021-03-18:
Mostly done, except for a couple build/CI steps I need help with.  See PR #94.

closed 2021-03-18 by henrygab