adafruit/tinyuf2 #95: mimxrt: Add `nuke.bin` (and .uf2?)

jepler, 2021-03-18:
@ladyada would like a "nuke" program for this platform.  It would just issue the general flash erase command, so that uf2 would have to be re-loaded via sdphost or j-link.

nxf58843, 2021-03-18:
The code just looks for the FCFB tag to trigger a reload.  Just take a file with 0x404 bytes of 0 and create a .UF2 file starting from 0x0.  
Using a 0x404 long file will make it universal since some parts look for the tag at 0x0 and some at 0x400.

ladyada, 2021-03-18:
yah what we want is particularly special - a RAM-loading bin that sdphost can load in that will call chip erase. there are some times you really just want to start over from the beginning :)

nxf58843, 2021-03-18:
All the pieces to do that are in the existing code.  Let me know if you need help.

ladyada, 2021-03-19:
@hathach please make an attempt and we'll ask @nxf58843 for help if we need advice :)

hathach, 2021-03-19:
the nuke binary is an application, doesn't sound like part of the bootloader project. Though I will try to fit it :sweat_smile: sweat_smile:

nxf58843, 2021-03-19:
I've been noodling on this and I think we can add this function to the same standard bootloader image so that you can make it erase or force an overwrite when you load it.  We can add a tiny image that sets a flag and then jumps to the main image so you can pick the behavior by the jump address.  Or, has anyone considered adding USB CDC commands for erasing?  That would be more cross platform.

hathach, 2021-03-19:
@nxf58843 thanks, regardless, I have added the PR for this https://github.com/adafruit/tinyuf2/pull/101 . Although it is technically not bootloader, but it is nice to have. I plan to generalize it so basically we will have 3 buidable binaries in this repo
- tinyuf2 bootloader (itself)
- self-udpate: for iMX RT it is actually the tinyuf2 itself. For other ports, it is an application than update the bootloader
- erase-app / nuke: to erase application firmware flash (factory reset), or even better if possible an additional nuke (erase-all) for imxrt

hathach, 2021-03-19:
update: trigger it via USB Control message is nice as well, though It will be more problematic to get it done in the host than firmware. With the spaghetti of driver installing, claiming interface on wins/linux/macos etc ... It is just much easier to drag an pre-built uf2 to do the work in my experience.

nxf58843, 2021-03-19:
sdphost also has a write register command.  We could use that to set flags that configure the behavior before jumping.  That would be simpler and save generating multiple images.  We could also specify what to erase (bl, app, everything) that way.  Only check the flags for sdp mode.

hathach, 2021-03-19:
hmm, maybe we can have an magic number (similar to double tap), so that application can perform the factory reset by itself :thinking: :thinking:  seem like there is several ways to implement this. Of course, we can implement it all, just make sure we document it well :)

hathach, 2021-03-19:
> sdphost also has a write register command. We could use that to set flags that configure the behavior before jumping. That would be simpler and save generating multiple images. We could also specify what to erase (bl, app, everything) that way. Only check the flags for sdp mode.

Yeah, I just think of this before seeing your reply. Therefore both app and sdphost can "execute" it. Also I was thinking for all other ports as well :)

nxf58843, 2021-03-19:
We could probably even create a .UF2 file that writes the magic number to trigger it through MSD.

hathach, 2021-03-19:
Right, nice suggestion, I will gather these all and put it into a pr

PR #102 (2021-03-23): Add Imxrt erase firmware
This PR fix #95  
- ground work for `erase_firmware-BOARD.uf2` for all ports
- new board API: board_reset() and board_flash_erase_app()
- add erase_firmware as std application that reset board with MAGIC ERASE_APP
- IMXRT implement board_flash_erase_app() add makefile to support erase_firmware. For iMXRT it is much quicker to perform chip erase and re-write bootloader from sram to flash. (sector/block erase probably takes longer).
- `erase_firmware-BOARD.uf2` is also uploaded as ci artifact/release if available.
- Also update tinyusb to latest as well (why not).
closed 2021-03-23 by hathach