adafruit/tinyuf2 #119: Issues with block size speedup of copying on ESP32-S2

UnexpectedMaker, 2021-05-26:
While the increased speed of copying is a really welcome change in 4.0, I am experiencing boards (FeatherS2 and TinyS2) that don't copy properly and don't re-mount afterwards - both flashing/copying on macOS and on raspberrypi os.

Often a board that fails multiple times in a row with v4 can be fixed by erasing it's flash and then a re-flash with v3 UF2, and then it works ok again. Note, erasing the flash and re-installing v4 doesn't work on these boards, but going back to v3 does.

So far I've had this with 6 of the first 20 FeatherS2 boards I've tried v4 on and 11 of the first 50 TinyS2s.  

So this is an issue with both internal and external Flash as the TinyS2 uses the ESP32-FN4R2 with internal Flash and PSRAM.

I've had to move my test jigs back to flashing with v3 now and I am back on 100% reliability again.

Maybe it's just too aggressive? Or has issues with QIO or flash speed?

hathach, 2021-05-27:
to be honest, I am not entirely sure, I only test it on the module wroom/wrover, It is probably issue with block erase command. Here is the pr https://github.com/adafruit/tinyuf2/pull/86/files . maybe your flash device need a bit of delay or something. I think the solution for now is adding an flag in your board to skip the block erase and keep using sector erase command so that it could move along with other feature. Feel free to tag me for review when making PR for thi.

UnexpectedMaker, 2021-06-03:
> block erase

I'd had a look through that PR, but really don't understand it. Def not something I can see my self reverting, sorry.

Can I just change the FLASH_CACHE_SIZE back to 4096?

I don't see how using an ESP32-S2 WROOM/WROVER would be any different to the chips I'm using... My FeatherS2 uses the same S2 & Flash as the WROVER. 

How many modules have you tested on? As I mentioned It's failing on around 20-30% - so unless you've tested this on enough modules you could easily be missing the issue.

UnexpectedMaker, 2021-06-03:
I also just tried creating a branch of release 0.3.0 - to build my TinyS2 UF2 boot loader from that, but I couldn't complete the submodule - keeps failing on nxp, and my git fu is just not good enough to solve why.

So basically I cant deploy CircuitPython on my TinyS2 boards via UF2 - super sad :(

Any guidance you can give me to how to specifically revert the 0.4.0 changes for ESP32-S2 so I can make a working UF2 boot loader, I'd really appreciate it.

hathach, 2021-06-03:
> Can I just change the FLASH_CACHE_SIZE back to 4096?

Possibly, I did a quick look, should work. please try. Ultimately, you need to add an board specific flag that defined in your board.h to overwrite the default one. Let just say `USE_FLASH_SECTOR_ERASE` or something similar. Then you could use that define to change the FLASH_CACHE_SIZE to 4096 in the board_flash.

UnexpectedMaker, 2021-07-11:
Hi folks, I realised I never reported back here - super sorry.
So I did have to set my cache size back to 4096 - but that wasn't enough, It was still failing 1 out of every 6-7 times on both my FeatherS2 and TinyS2 and what I found was that I also had to wait 3-5 second's after I copied the CPY.uf2 firmware on, even after it `finished copying` before I could allow it to remount and that brought me back to 100% reliability.

Because of this pause that I have to do on my test jigs, and no one in the wild that I can see reporting issues with larger block sizes copying from their computers, I decided to not make any changes to the repo for my boards, and plan to re-assess it if/when a 0.5.0 comes out.

So I'm closing this `finally` and I'll report back in the future if I have further issues.

Cheers :)

closed 2021-07-11 by UnexpectedMaker

hathach, 2021-07-11:
You should wait for the board to do its own self reset. Else you could reset it while writing the last block. `finished copying` is to the pc side only. Glad you work it out.