adafruit/tinyuf2 #131: Help with mounting errors on GigaDevice GD32VF103 port

KarlK90, 2021-07-09:
I started porting tinyuf2 (great project!) to the RISC-V GigaDevice GD32VF103 on the Sipeed Longan Nano Board, which has supposedly the same USB OTG peripheral as the STM32F105/107 or STM32F4s. Flash writing and jumping to the application is not possible ATM, but that should be easy to implement. Now to my problem:

So far I got it to boot and enumerate successfully. But it crashes the moment that it tries to mount the ghost fast partition, already the first READ10 command of the boot block never succeeds. I have attached an dmesg dump of the problem, I can also post the debug log of `LOG=2` if this is any help. When stepping into the code with a debugger it sometimes hits this [breakpoint](https://github.com/KarlK90/tinyusb/blob/gd32vf103-support-tiny-usb/src/portable/st/synopsys/dcd_synopsys.c#L953). Unfortunately I have limited experienced when it comes to debugging USB and I'm a bit lost where to look for the root cause, that's why I'm asking directly here. (although a logic analyzer and oscilloscope is at hand).

I have uploaded the [current state of affairs in my personal fork](https://github.com/KarlK90/tinyuf2/tree/gd32vf103-support), which should be easily compileable e.g. with an [embecosm toolchain](https://www.embecosm.com/resources/tool-chain-downloads/#riscv-stable) if somebody want to have a look.

<details>
<summary>DMESG output</summary>
```
[ 5575.780639] usb 1-1.4.2: new full-speed USB device number 50 using xhci_hcd
[ 5576.126100] usb 1-1.4.2: New USB device found, idVendor=239a, idProduct=0069, bcdDevice= 1.00
[ 5576.126104] usb 1-1.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5576.126106] usb 1-1.4.2: Product: LonganNano
[ 5576.126107] usb 1-1.4.2: Manufacturer: GD
[ 5576.126108] usb 1-1.4.2: SerialNumber: 56F34235363501000055396D
[ 5576.169130] usb-storage 1-1.4.2:1.0: USB Mass Storage device detected
[ 5576.169267] scsi host8: usb-storage 1-1.4.2:1.0
[ 5576.199265] hid-generic 0003:239A:0069.0010: hiddev1,hidraw3: USB HID v1.11 Device [GD LonganNano] on usb-0000:03:00.0-1.4.2/input1
[ 5577.184461] scsi host8: scsi scan: INQUIRY result too short (5), using 36
[ 5577.184465] scsi 8:0:0:0: Direct-Access     Adafruit UF2 Bootloader   1.0  PQ: 0 ANSI: 2
[ 5577.184630] sd 8:0:0:0: Attached scsi generic sg2 type 0
[ 5577.184972] sd 8:0:0:0: [sdc] 65801 512-byte logical blocks: (33.7 MB/32.1 MiB)
[ 5577.185144] sd 8:0:0:0: [sdc] Write Protect is off
[ 5577.185145] sd 8:0:0:0: [sdc] Mode Sense: 03 00 00 00
[ 5577.185324] sd 8:0:0:0: [sdc] No Caching mode page found
[ 5577.185326] sd 8:0:0:0: [sdc] Assuming drive cache: write through
[ 5577.283653] usb 1-1.4.2: reset full-speed USB device number 50 using xhci_hcd
[ 5577.603637] usb 1-1.4.2: reset full-speed USB device number 50 using xhci_hcd
[ 5586.104818] usb 1-1.3: Failed to suspend device, error -110
[ 5592.976398] usb 1-1.4.2: device descriptor read/64, error -110
[ 5608.660128] usb 1-1.4.2: reset full-speed USB device number 50 using xhci_hcd
[ 5624.083872] usb 1-1.4.2: device descriptor read/64, error -110
[ 5626.983180] usb 1-1.4.2: USB disconnect, device number 50
[ 5626.987785] print_req_error: 33 callbacks suppressed
[ 5626.987786] blk_update_request: I/O error, dev sdc, sector 1 op 0x0:(READ) flags 0x0 phys_seg 7 prio class 0
[ 5626.987791] buffer_io_error: 75 callbacks suppressed
[ 5626.987792] Buffer I/O error on dev sdc, logical block 1, async page read
[ 5626.987795] Buffer I/O error on dev sdc, logical block 2, async page read
[ 5626.987796] Buffer I/O error on dev sdc, logical block 3, async page read
[ 5626.987797] Buffer I/O error on dev sdc, logical block 4, async page read
[ 5626.987798] Buffer I/O error on dev sdc, logical block 5, async page read
[ 5626.987799] Buffer I/O error on dev sdc, logical block 6, async page read
[ 5626.987800] Buffer I/O error on dev sdc, logical block 7, async page read
[ 5626.987963] blk_update_request: I/O error, dev sdc, sector 1 op 0x0:(READ) flags 0x0 phys_seg 7 prio class 0
[ 5626.987967] Buffer I/O error on dev sdc, logical block 1, async page read
[ 5626.987968] Buffer I/O error on dev sdc, logical block 2, async page read
[ 5626.987970] Buffer I/O error on dev sdc, logical block 3, async page read
[ 5626.987979] blk_update_request: I/O error, dev sdc, sector 1 op 0x0:(READ) flags 0x0 phys_seg 7 prio class 0
[ 5626.987984] ldm_validate_partition_table(): Disk read failed.
[ 5626.987989] blk_update_request: I/O error, dev sdc, sector 1 op 0x0:(READ) flags 0x0 phys_seg 7 prio class 0
[ 5626.987997] blk_update_request: I/O error, dev sdc, sector 1 op 0x0:(READ) flags 0x0 phys_seg 7 prio class 0
[ 5626.988004] blk_update_request: I/O error, dev sdc, sector 1 op 0x0:(READ) flags 0x0 phys_seg 7 prio class 0
[ 5626.988012] blk_update_request: I/O error, dev sdc, sector 1 op 0x0:(READ) flags 0x0 phys_seg 7 prio class 0
[ 5626.988016] Dev sdc: unable to read RDB block 0
[ 5626.988021] blk_update_request: I/O error, dev sdc, sector 1 op 0x0:(READ) flags 0x0 phys_seg 7 prio class 0
[ 5626.988027] blk_update_request: I/O error, dev sdc, sector 1 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[ 5626.988031] blk_update_request: I/O error, dev sdc, sector 2 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[ 5626.988106]  sdc: unable to read partition table
[ 5626.988352] sd 8:0:0:0: [sdc] Read Capacity(10) failed: Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK
[ 5626.988355] sd 8:0:0:0: [sdc] Sense not available.
[ 5626.988357] sd 8:0:0:0: [sdc] 0 512-byte logical blocks: (0 B/0 B)
[ 5626.988363] sdc: detected capacity change from 65801 to 0
[ 5626.988365] sd 8:0:0:0: [sdc] Attached SCSI removable disk
[ 5628.265768] usb 1-1.4.2: new full-speed USB device number 5
```
</details>

hathach, 2021-07-10:
you should try to get the stack debug log with `LOG=2` to see at which command it failed. Since GD32Vf is a new port, it would be easier if you test it with https://github.com/hathach/tinyusb first to see if all stock example work to verify it is not the tinyusb stack issue. I think I have this GD32VF103 somewhere in my drawer as well, if you submit an PR to add BSP for it and, I could help to troubleshoot if it is indeed usb stack issue.

KarlK90, 2021-07-10:
Thank you very much :slightly_smiling_face:! I have openend a draft PR at tinyusb. I have run the tinyuf2 again with `LOG=2` and captured the output of two runs that produced slightly different points where it fails: 

**tinyutf2**

[log_hid.txt](https://github.com/adafruit/tinyuf2/files/6795066/log_hid.txt)
[log_scsi_read10.txt](https://github.com/adafruit/tinyuf2/files/6795067/log_scsi_read10.txt)

**tinyusb examples**

This is an output of the cdc_msc example, that seems to work fine (?):

[log_demo_csc_msc.txt](https://github.com/adafruit/tinyuf2/files/6795073/log_demo_csc_msc.txt)

The cdc dual ports example fails with an assert though:

[log_demo_csc_dual_ports.txt](https://github.com/adafruit/tinyuf2/files/6795085/log_demo_csc_dual_ports.txt)

hathach, 2022-07-11:
@KarlK90 would you mind confirming if we could close this ?

KarlK90, 2022-07-11:
@hathach I'll finally open a PR to support tinyuf2 on GD32VF103 in the next days, we could close it afterwards?

hathach, 2022-07-11:
sure things :+1: