adafruit/tinyuf2 #134: ESP32S2 self-update not working

T94T, 2021-07-21:
We have two different self-update UF2 firmware files. But uploading one or another file does not work. The UF2 bootloader that was last burned via the `idf.py` tool is always active and cannot be overwritten.

hathach, 2021-07-21:
can you provide more information on the issue:
1. which board you are running
2. step to reproduce
3. all the version of firmware, software that are in used.

T94T, 2021-07-22:
1. ESP32-S2-Saola-1RI
2./3. We are using latest https://github.com/adafruit/tinyuf2/commit/7b92803108d77e94ada567fd9fbaf496256984eb and building with `make BOARD=espressif_saola_1_wroom all self-update`. The generated `update-tinyuf2.uf2`can be uploaded to the board but it does not take effect.

hathach, 2021-07-22:
2.Step to reproduce: please provide your procedure detail, what have you done  to test and how you verify that it does not take effect.

T94T, 2021-07-22:
Just started all over again with following steps:
- `git clone --recurse-submodules https://github.com/adafruit/tinyuf2`
- `install.sh`
- `cd tinyuf2/ports/esp32s2`
- `get_idf`
- `idf.py -p /dev/ttyUSB0 -DBOARD=espressif_saola_1_wroom flash`
After this last step I have the UF2 bootloader correctly burned to the module.

Then I created the `update-tinyuf2.uf2` for backup:
- `make BOARD=espressif_saola_1_wroom all self-update`

To check wheter self-update works or not I created a new `update-tinyuf2.uf2` file with `UF2_VOLUME_LABEL` changed from default "`SAOLA1MBOOT`" to "`UPDATE`".
- `make BOARD=espressif_saola_1_wroom clean`
- `make BOARD=espressif_saola_1_wroom all self-update`

And **now it works as expected**. After a self-update the MSD volume name is `UPDATE`. Also reverting back to the `UF2_VOLUME_LABEL` version works. Very strange. Must have  messed up something with my previous installation.

hathach, 2021-07-22:
update-uf2 is an actually an app like circuit python, it covert tinyuf2.bin to carray then erase and write once when updating. Maybe you modify but didn't recompile the tinyuf2 first (all target) and updater use the old bin file which make you think it does not do the work.

hathach, 2021-07-22:
Ok now I see what you mean, the changes you did isn't part of the tinyuf2, it is stage2 bootlaoder, tinyuf2 is actually factory app. Right now the updater only update tinyuf2 since I am not sure if it is safe and/or possible to update stage2 bootloader. I didn't say it is not possible, I just haven't tried it yet.

T94T, 2021-07-22:
Thank you very much for investigating your time! Now some things become clear. Closing for now...

closed 2021-07-22 by T94T

hathach, 2021-07-22:
If you still want to update the boot2 as well. You could open an issue for it, I will try to see if that is possible to do when having time

hathach, 2021-07-23:
I tried to update the boot2 binary with update-uf2, there is a couple of issue
1. There is a macro check to validate the address, though it is easily by passed by enabling unsafed write https://github.com/espressif/esp-idf/blob/master/components/spi_flash/esp_flash_api.c#L51
2. There is some meta data required for boot2 bin
```
Invalid image block, can't boot.
ets_main.c 386 
```
Digging to esptool.py https://github.com/espressif/esptool/blob/master/esptool.py#L3488, look like it does some modification to bootlaoder image to add image size, flash mode, flash freq etc to the very frist 4 bytes (header). At this point, I don't think it is the worth effort to update it, since it seem rather unsafe. It is rather easy to flash boot2 update via rom bootloader. Boot2 won't change much anyway.

T94T, 2021-07-23:
Thank you for digging even deeper. "unsafe" does not sound good. We are now flashing the boards directly with the upload tool. The failover bootloader we only need during the development phase. So it is not a big problem. Thank you very much again.

hathach, 2021-07-23:
after giving a bit more thought, I think it is still worth updating, since ROM loader is there to unbrick. However, I don't quite feel comfortable following the esptool code, maybe if someone know where is the specs for the layout. We could get this done. I push my interim work to the branch here https://github.com/adafruit/tinyuf2/tree/esp32s2-update-boot2 . If anyone could help with meta data, that would nail it.