adafruit/tinyuf2 #136: Error when installing CircuitPython on a Feather STM32F405 Express using TinyUF2

cdwilson, 2021-07-26:
### Operating System

MacOS

### Board

Feather STM32F405 Express

### Firmware/Software/Tool Version

TinyUF2 (0.5.0 release): https://github.com/adafruit/tinyuf2/releases/download/0.5.0/tinyuf2-feather_stm32f405_express-0.5.0.zip
CircuitPython (`main` branch): adafruit/circuitpython@b9fa06cf6bb3c45eaf57724d351a04564d562b6a

### What happened ?

I am trying to install CircuitPython on a Feather STM32F405 Express using the TinyUF2 bootloader.  When I try to install the CircuitPython firmware by copying the `firmware.uf2` image onto the `FTHR405BOOT` mass storage device, it starts to copy the firmware:

[image]

However, about 5-10s later, I get the following error:

[image]

The CircuitPython image does not run (the board is unresponsive) and pressing the reset button bricks the board (must be recovered with DFU).

### How to reproduce ?

1. Install UF2 bootloader on the Feather STM32F405 Express:
```
dfu-util -a 0 --dfuse-address 0x08000000 -D ~/Downloads/tinyuf2-feather_stm32f405_express-0.5.0/tinyuf2-feather_stm32f405_express-0.5.0.bin
```
2. Build circuitpython image with UF2 bootloader support
```
cd circuitpython
git checkout b9fa06cf6bb3c45eaf57724d351a04564d562b6a
make -C ports/stm/ BOARD=feather_stm32f405_express UF2_BOOTLOADER=1
```
3. Copy `build-feather_stm32f405_express/firmware.uf2` to `/Volumes/FTHR405BOOT/` disk
4. Wait for error message to occur

cdwilson, 2021-07-26:
FYI, if I program the UF2 bootloader and the CircuitPython `firmware.bin` firmware images manually via DFU, the board boots the CircuitPython image without issues:

```
dfu-util -a 0 --dfuse-address 0x08000000 -D ~/Downloads/tinyuf2-feather_stm32f405_express-0.5.0/tinyuf2-feather_stm32f405_express-0.5.0.bin

dfu-util -a 0 --dfuse-address 0x8010000 -D ports/stm/build-feather_stm32f405_express/firmware.bin
```

[image]

hathach, 2021-07-26:
hmm, how did you build the circuitpython for f4, You will need to run make and specify `UF2_BOOTLAODER=1` to select the linker with bootloader in mind. default linker is no bootloader. The main reason for this, is tinyuf2 project started much later than the circuittpython port of stm. Let me know if you still have issue with it.

```
make BOARD=feather_stm32f405_express UF2_BOOTLOADER=1
```

reppad, 2021-07-27:
I'm not sure if this will help but here is a copy/paste of my [recent message on adafuit/tinyuf2 discord channel](https://discord.com/channels/327254708534116352/807372144631742484/866603652302241832), the behavior might be related.

> I would like some advice for using the tinyuf2 bootloader on a STM32F401CCU6 Blackpill module.
> I flashed the tinyuf2-stm32f401_blackpill-0.5.0 firmware on the module, once plugged in I have a "BlackPill" drive which contains "current.uf2", "index.html" and "info_uf2.txt", so everything looks good.
> 
> I tried to convert a bin file to UF2 format (the orginal bin works fine when it is flashed directly to address 0x08000000 without the bootlader), for that I used the following command (uf2conv is a Rust crate):
> uf2conv firmware.bin -b 0x08010000 -f 0x57755a57
> I also tried with the uf2conv.py script provided by Microsoft on the microsoft/uf2 repo:
> uf2conv.py firmware.bin -c -b 0x08010000 -f 0x57755a57
> 
> I got the chip family id and base address directly from the tinyuf2 port code to be (a bit more) sure of what values ​​to use.
> 
> When I copy the generated uf2 file into the bootloader drive, the drive disconnects (looks normal) but instead of booting on my program the bootloader restarts.
> I have tried several base address values ​​and the only difference is that sometimes the bootloader cannot be restarted (not surprisingly in case I overwrite the bootloader).
> 
> For information I use the same UF2 conversion method without problem on Adafruit SAMD21/51 modules.

hathach, 2021-07-29:
I am able to reproduce the issue, looking into it now.

closed 2021-07-29 by hathach

hathach, 2021-07-29:
should be fixed by #137, please try again

cdwilson, 2021-07-31:
Tried again with #137 and everything appears to be working correctly.  Thanks @hathach!

hathach, 2021-07-31:
Thanks for confirmation, It is released as 0.5.1

PR #137 (2021-07-29): fix updating issue with circuipython due to large size
should fix #136  take the chance to clean up flashing code. may also fix (or not) #75