adafruit/tinyuf2 #147: fail using update-tinyuf2-lolin_s2_mini-0.5.2.uf2

durapensa, 2021-09-08:
### Operating System

Linux

### INFO_UF2.TXT

```
TinyUF2 Bootloader 0.5.2 - tinyusb (0.10.1-109-g9b3ec69b)
Model: Lolin S2 Mini
Board-ID: ESP32S2FN4R2-S2Mini-1-0-0
Date: Sep  8 2021
```

### What happened ?

When writing `update-tinyuf2-lolin_s2_mini-0.5.2.uf2`, OS reports WRITE I/O error, seemingly because of USB disconnect.

Board resets back to UF2 bootloader mode, with previously flashed firmware gone (`user_fs` FAT partitions seems okay).

Related #138 ?

### How to reproduce ?

1. Enter UF2 bootloader mode, `S2MINIBOOT` drive appears.
2. Copy `update-tinyuf2-lolin_s2_mini-0.5.2.uf2` to `S2MINIBOOT` 

### Debug Log

Flashed build with LOG=3, but board has no UART pins and USB CDC not enabled for onboard UART passthrough.

hathach, 2021-09-09:
Regardless of the os error, was the upgrade complete successfully ?

hathach, 2022-06-17:
closed due to lack of response

closed 2022-06-17 by hathach