adafruit/tinyuf2 #149: Move to IDF 4.4 for ESP32-S3 support

UnexpectedMaker, 2021-09-15:
Hi @hathach :)

I'm working on ESP32-S3 support for CPY and that means we'll also need support for S3 in TinyUF2 and that requires being on the latest master IDF branch - 4.4.bleeeeeeeeeding edge, and we'll also need to stay updated often.

I'm happy to do the work on this, but I wanted to check it's ok before I start as it *may* need a substantial change like CPY needed - where we seperate esp32s2 and esp32s3 stuff into partitioned folders.

So can you all let me know if there is a preference on how you want this done? Or can I do what I did in CPY and run it all from Makefile variables and skconfig definitions?

Thanks!

hathach, 2021-09-17:
> Hi @hathach :)
> 
> I'm working on ESP32-S3 support for CPY and that means we'll also need support for S3 in TinyUF2 and that requires being on the latest master IDF branch - 4.4.bleeeeeeeeeding edge, and we'll also need to stay updated often.

That is great to hear, which branch/tag IDF you plan to make an upgrade, there is an new 4.3.1 which is released just a week ago. Once tested locally, please make PR to update IDF as standalone, it help to get it merged quickly.

> I'm happy to do the work on this, but I wanted to check it's ok before I start as it _may_ need a substantial change like CPY needed - where we seperate esp32s2 and esp32s3 stuff into partitioned [...]

For folder changed, there is 2 ways we could go
1. If s2 vs s3 changes is small enough we could have them under the same `ports/esp32sx`
2. If changes are too different we can have them as separated port esp32s2 and esp32s3 respectively (maybe under ports/espressif. Then we can move common code in esp32s2/components in ports/espressif as well where both ports can refer to minimize duplication

IMHO, I would prefer to have s2 and s3 has their own port folder, it helps with managing and adding new boards easier and more explicit. We only need to make sure to not introduce code duplication which make maintaining difficult in the future.

Currently I don't have any S3 boards, and don't have anyway to obtain it under my city strict lockdown. Therefore my help would be a bit limited.

UnexpectedMaker, 2021-09-17:
> That is great to hear, which branch/tag IDF you plan to make an upgrade, there is an new 4.3.1 which is released just a week ago. Once tested locally, please make PR to update IDF as standalone, it help to get it merged quickly.
> 

It has to be latest master (4.4-dev), and it will need to kept updated constantly as work on the S3 IDF stuff continues. 
S2 is also working fine in 4.4-dev - The CPY work I've done moved S2 to IDF 4.4-dev as well.
 
> For folder changed, there is 2 ways we could go
> 
> 1. If s2 vs s3 changes is small enough we could have them under the same `ports/esp32sx`

The differences are small enough in this case - making 2 seperate ports would 100% duplicate stacks of code for no reason.

> 2. If changes are too different we can have them as separated port esp32s2 and esp32s3 respectively (maybe under ports/espressif. Then we can move common code in esp32s2/components in [...]
Having them be their own ports will only introduce code duplication as they are identical. I don't think they need their own ports. It was not needed for CPY, but it's your repo, so I'm happy to do it how you want, but it will just be a duplicate of the S2 folder.

> Currently I don't have any S3 boards, and don't have anyway to obtain it under my city strict lockdown. Therefore my help would be a bit limited.

Mouser/Digikey might have a small number of Espressif S3 dev kits in stock soon, but production silicon is not slated to start until December. 

I'll try to get something together as soon as possible and do a compare for you to see what I have in mind :)

hathach, 2021-09-17:
> It has to be latest master (4.4-dev), and it will need to kept updated constantly as work on the S3 IDF stuff continues.
S2 is also working fine in 4.4-dev - The CPY work I've done moved S2 to IDF 4.4-dev as well.

I have no issue with latest maser, though I still prefer to stick with tag/release since bootloader doesn't use much from IDF, mostly its peripherals and soc driver only. Anyway, please do update to whatever version that would work, eventual we will have an release/tag that work with S2 & S3 and settled then.

> The differences are small enough in this case - making 2 seperate ports would 100% duplicate stacks of code for no reason.

As mentioned above, If the changes are small enough, we could pack them together. Please go ahead with the changes that you thing work best. We could always adjust it later on if needed. A few month ago espressif make PR to tinyusb to add esp32s3, they fisrt use 1st approach then changed to 2nd approach by their own. We will see if that is still the case. https://github.com/hathach/tinyusb/pull/783

s-light, 2021-11-03:
a little off-topic:
> A boards I ordered 3 months ago from digikey is still "waiting for clearance" for more than 2 month. Hopefully the shipping will be better in the next couple of months.

i just [checked stock at mouser](https://www.mouser.de/c/?q=ESP32-S3-DevKitC-1-N8) and *now* they have ~380 `ESP32-S3-DevKitC-1-N8` in stock.

hathach, 2021-11-05:
thank you for your info. I already back-order an couple of s3 board from digikey. It will be delivered at the end of November if they estimate it correctly.

tannewt, 2021-11-19:
The move to 4.4 was done in https://github.com/adafruit/tinyuf2/pull/156

closed 2021-11-19 by tannewt