adafruit/tinyuf2 #163: TinyUF2 is missing for Lolin S2 Pico in release 0.5.2

geekmomprojects, 2021-12-08:
### Operating System

Windows 10

### INFO_UF2.TXT

N/A

### What happened ?

The "download bootloader zip" link from the circuitpython.org board page for the Lolin Pico S2 (https://circuitpython.org/board/lolin_s2_pico/) indicates there should be a bin file for the board in the latest release, but the bin file does not exist and the link is broken.

### How to reproduce ?

Go to https://circuitpython.org/board/lolin_s2_pico/
Click the "Download Bootloader Zip" button
The bin file is "not found"

hathach, 2021-12-08:
the board is added after latest release of 0.5.2. It will be available in the next release.

hathach, 2021-12-08:
should be available in 0.6.0

closed 2021-12-08 by hathach