adafruit/tinyuf2 #165: STM32F4 is not building for gcc-11

DuMaM, 2021-12-11:
### Operating System

Linux

### INFO_UF2.TXT

There is no need of info_uf2.txt file.

### What happened ?

I'm unable to build current release for stm32f4, also there is no update img in github release.

### How to reproduce ?

1. Make sure you are up to date with master branch and submodules
2. Check if your gcc version is:
``` shell
arm-none-eabi-gcc (Arch Repository) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
3. Go to ports/stm32f4
3. Run `make BOARD=stm32f411ce_blackpill all`

### Debug Log

```In function 'board_usb_get_serial',
    inlined from 'tud_descriptor_string_cb' at /home/dumam/Projects/tinyuf2/src/usb_descriptors.c:190:20,
    inlined from 'process_get_descriptor' at /home/dumam/Projects/tinyuf2/lib/tinyusb/src/device/usbd.c:984:50,
    inlined from 'process_control_request' at /home/dumam/Projects/tinyuf2/lib/tinyusb/src/device/usbd.c:680:11,
    inlined from 'tud_task' at /home/dumam/Projects/tinyuf2/lib/tinyusb/src/device/usbd.c:522:15,
    inlined from 'tud_task' at /home/dumam/Projects/tinyuf2/lib/tinyusb/src/device/usbd.c:474:6,
    inlined from 'main' at /home/dumam/Projects/tinyuf2/src/main.c:92:5:
/home/dumam/Projects/tinyuf2/ports/stm32f4/boards.c:241:3: error: 'memcpy' reading 12 bytes from a region of size 0 [-Werror=stringop-overread]
  241 |   memcpy(serial_id, STM32_UUID, len);
      |   ^
compilation terminated due to -Wfatal-errors.
lto1: all warnings being treated as errors
lto-wrapper: fatal error: arm-none-eabi-gcc returned 1 exit status
compilation terminated.
/usr/lib/gcc/arm-none-eabi/11.2.0/../../../../arm-none-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [../rules.mk:44: _build/stm32f411ce_blackpill/tinyuf2-stm32f411ce_blackpill.elf] Błąd 1
```

hathach, 2021-12-25:
I couldn't reproduce the issue since latest release from ARM is only 10.3. We will leave this issue and pr pending until gcc11 is released. Thank you for the issue.

DuMaM, 2021-12-28:
Oh... I need to take a dipper look on this package then.
https://archlinux.org/packages/community/x86_64/arm-none-eabi-gcc/

hathach, 2021-12-28:
> Oh... I need to take a dipper look on this package then. https://archlinux.org/packages/community/x86_64/arm-none-eabi-gcc/

yeah, Arch Linux always use latest software. I am sure it is part of gcc11 new feature, it will take a bit of time for ARM to release gcc11 for me to try it out. Since this warning is new to me, I would like to try it out to see if there is any other way to fix this.

tannewt, 2021-12-28:
I also use Arch and have seen this too. There is a bug asking for the suggested fix here: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578#c16

Here is how I did it [in CP](
https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/nvm/ByteArray.c#L68-L76):
```c
    #pragma GCC diagnostic push
    #if __GNUC__ >= 11
    // TODO: Update this to a better workaround for GCC 11 when one is provided.
    // See: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578#c20
    #pragma GCC diagnostic ignored "-Warray-bounds"
    #pragma GCC diagnostic ignored "-Wstringop-overread"
    #endif
    memcpy(buffer, (uint8_t *)CIRCUITPY_INTERNAL_NVM_START_ADDR, FLASH_SECTOR_SIZE);
    #pragma GCC diagnostic pop
```

hathach, 2021-12-29:
thanks @tannewt for more input, it still looks a bit odd to me. I will look more into this once ARM officially release gcc11.

DuMaM, 2022-04-28:
there was update of toolchain
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads-1

hathach, 2022-07-01:
fixed by #166

closed 2022-07-01 by hathach