adafruit/tinyuf2 #176: PIN_DOUBLE_RESET_RC on ESP32-S3 - using IO47 not working

UnexpectedMaker, 2021-12-27:
### Operating System

MacOS

### INFO_UF2.TXT

TinyUF2 Bootloader 0.6.2-29-g06a65c8 - esp-idf (v4.4-dev-3608-gbbe2a1bf34) tinyusb (0.12.0-203-ga4cfd1c6)
Model: Unexpected Maker TinyS3
Board-ID: ESP32S3-TinyS3-01
Date: Dec 27 2021

### What happened ?

I'm using IO47 for my PIN_DOUBLE_RESET_RC and it's not working and I'm not sure why.

Datasheet shows that it's Reset mode is E1 (Input enabled) but it's default function is (F0) - SPICLK_P_DIFF which is O/T, not GPIO47 which is I/O/T - (F1)

I've confirmed I have plenty of hold high time on the IO during the boot cycle.  I've even gone to a 2.2uF cap with 1Mohm resistor, and I can see it's holding 3V+ for ages - way longer than I need... but for whatever reason, it never catches it as high and just moves on to load into CPY.

I've confirmed it running through the correct code as I have the LED going blue instead of purple if PIN_DOUBLE_RESET_RC code is being run, and everything looks correct.

I can't see anything in the TinyUF2 or IDF code that would prevent IO47/48 from being used as a GPIO - they are both new and exclusive to the S3, but the only difference is the F0/F1 states.

I've even tried to brute force it...
```
esp_rom_gpio_pad_select_gpio(PIN_DOUBLE_RESET_RC);
gpio_ll_output_disable(&GPIO, PIN_DOUBLE_RESET_RC);
gpio_ll_input_enable(&GPIO, PIN_DOUBLE_RESET_RC);
```

I don't have access to `driver/gpio.h` in `bootloader-start.c` so I was unable to try `gpio_config` to set the GPIO to Input.

Any ideas what I can do?   IO47/48 are the only IO I have spare that I can use for PIN_DOUBLE_RESET_RC so I need to get this working.

Thanks!

### How to reproduce ?

Not sure how you can repo this.

ladyada, 2021-12-27:
the JTAG pins are special, they cannot be used for the RC double-click because the boot in JTAG mode, i think pins 38 and below are ok

ladyada, 2021-12-27:
ok! thats something that has bit me before is all :)
when @hathach is back from holiday they can look in more detail

hathach, 2021-12-27:
since we don't have your TinyS3, please try to test/confirm it with S3 devkitC
1. Tell us what is your R and C value, I will try to reproduce it here
2. confirm the issue occurred on S3 devkit
3. With the same RC setup. move to other non-JTAG pin and confirm that it works.
4. Screenshot with scope testing with (2) and (3) for reference would help

UnexpectedMaker, 2022-01-02:
Sorry for the delay on replying here... I spent 2 full days on this, and the results are not good news.

I cannot get this to work on any final production silicon I have here, on any board of mine, on any IO - no matter if the components are onboard or connected externally on a breadboard or soldered directly to the header pins.

**But...**   I can get it to work with any IO including IO47 on my S3-DevkitC on a breadboard... the thing is, my DevKitC is using pre-production silicon - not final production. The board doesn't have the RF shield on it, so I can confirm it's pre-production.

What I am seeing is that the cap is draining so fast through the IO once the high state is dropped to low, that it's lower than the threshold needed to read it as high.  Using the GPIO HAL, I cannot find any combination of pin hold, or pin state that doesn't drain the cap too much during the second reset press.

I even stuck a 10uF cap on there and held the high state on first reset for 1 second , to ensure it was charged, and even that drained too fast on the second reset to trigger a HIGH state when read back.

On my DevKit I am able to use a 400k resistor and 1uF Cap, and it works on any IO, though with the 1uF it's a little harder to get perfect timing for the second click, but definitely usable.

I don't have a newer "final" S3-DevKit here to test on, to confirm or refute my findings, but I assume one of you ( @ladyada or @hathach ) have a release S3 devkit you can try this on?

UnexpectedMaker, 2022-01-02:
Just some extra info on things I tried....
- I set the IO to Input instead of setting it to low at the end of the function, so try to prevent it from draining through the IO. Doesn't work as it seems resetting the S2 puts the IO into a specific starting state and it drains anyway.
- I've removed the discharge resistor completely, meaning the cap can only discharge back into the IO and I can confirm the cap stays charged until reset it pressed again, but instantly drops. 
- Tried lots of cap sizes, and holding charge state for various lengths in the function (charging a CAP is slower than discharging) to ensure it was at capacity before reset.

The only difference (apart from the silicon revision) between my boards and the S3 DevKit in terms of ESP-S3 specific HW design is that I am not using as much bulk capacitance and the inductor on the Analog power domain. Otherwise, it's the same.

I'm really hoping this is just me being silly and it's something simple I've overlooked, but at this stage, it has me stumped.

hathach, 2022-01-05:
I have re-tested with Adafruit S2 boards, all stills work well with latest code. 

[image] 
  
I will test this out on my S3 devkit. I have an working pair of through hole RC but it is hidden somewhere. Meanwhile, have you hooked up the scope to pin to see the state of the PIN_DOUBLE_RESET_RC. You could use that to adjust the RC value.

hathach, 2022-01-05:
I think I have narrowed the issue, it is probably due to the latest code. Doing more tests with scope hooked up, will post update whenever I got one.

closed 2022-01-05 by hathach

UnexpectedMaker, 2022-01-05:
This is still not working for me - at least not on IO47 on any of my boards, using a 1uF cap and 442K resistor :(

UnexpectedMaker, 2022-01-05:
Ok, I got it working on my FeatherS3 and TinyS3 by using a 2.2uF cap instead of the 1uF. Seems like the 1uF is still draining past the high voltage threshold too quickly.  Wicked!

PR #181 (2022-01-05): fix double tap issue
fix double tap issue, fix #176 . Root cause: `gpio_ll_set_level()` is used in version 0.7.0 since S3 gpio rom does not have `gpio_output_set_high()`, unlike rom API, gpio_ll needs to explicitly enable output first.

@ladyada while working on this issue, I figured out that `esp_rom_gpio_pad_select_gpio()` can also be used to get JTAG pin working as double tap pin. Therefore there is no restriction on JTAG pin after this got merged. Please skip 0.7.0 for new S2 board, 0.8.0 will be released as soon as this is merged.