adafruit/tinyuf2 #177: Board name inconsistency in tinyuf2 for the Lolin S2 Pico - causing templating problems on circuitpython.org

evildave666, 2021-12-29:
### Operating System

Others

### INFO_UF2.TXT

NA

### What happened ?

There is a board naming inconsistency between Circuitpython and tinyuf2 for the Lolin S2 Pico. This is causing the templating for the tinyuf2 download on circuitpython.org board page to be a dead link.

CP: lolin_s2_pico tinyuf2: lolin_s2-pico - I think tinyuf2 is the inconsistent one here in that other tinyuf2 board names use all underscores.

### How to reproduce ?

Go to https://circuitpython.org/board/lolin_s2_pico/ and click on the "Download Bootloader ZIP" button.

hathach, 2021-12-29:
@skieast what do you think, should we rename the board to match CP naming ?

skieast, 2021-12-29:
> @skieast what do you think, should we rename the board to match CP naming ?

Yes, I probably had some idea when i created the board but all the other ones use all underscores. Who knows. But should be changed for consistency.

PR #179 (2021-12-30): Rename board lolin_s2-pico to lolin-s2_pico (match CircuitPython board)
renamed board to match CircuitPython.  Hopefully done correctly. 
Closes #177
closed 2021-12-30 by hathach