adafruit/tinyuf2 #186: info_uf2.txt is empty in latest master

UnexpectedMaker, 2022-01-21:
### Operating System

MacOS

### INFO_UF2.TXT

empty

### What happened ?

just built latest from masterful my S3 boards, as a final test before I committed the to the repo, and though they work fine, the info_uf2.txt is empty.

### How to reproduce ?

Not sure how you can repo this.  I lent out my S3 devkit, so cant test it on that here, but it might also have the same issue.

henrygab, 2022-01-23:
Can you dump the full disk image (setor-by-sector ... on un*x-like systems, I think something like:

`dd if=/dev/hdX of=~/ghostfat_empty_uf2.img`, where you replace `/dev/hdX` with the ghostfat board device (different on each system).

If so, can you place in publicly accessible place?  Note that this file will include the device firmware....

henrygab, 2022-01-23:
Also, if you could also load a build from the `add-favicon-hg3` branch, and dump that full disk image, that'd be useful.

P.S. - love the Tiny Pico board you made.

hathach, 2022-01-25:
1. @UnexpectedMaker have you tried to plugged the board into other PC with other OS such as ubuntu/windows to make sure this is macOS specific issue ? I just checked with Ubuntu 20.04 and it still works. 
2. Which `BOARD=` you are using, make sure it is `espressif_esp32s3_devkitc_1` to rule out any chances that it is not your board porting issue.

UnexpectedMaker, 2022-02-07:
Oh, I totally forgot to reply to this....  Issue went away when you (@hathach ) reverted the favicon stuff you were playing with. Cant remember what exact commit I was on, but latest (what I submitted my ESP32-S3 boards agains) works perfectly, so closing this issue. Sorry!

closed 2022-02-07 by UnexpectedMaker

hathach, 2022-02-07:
I didn't reverted the favicon, it is already on the master branch, please double check it.

UnexpectedMaker, 2022-02-07:
Yes I know it's in there, but there was a sequence of commits you did and then fixed or changed. I can't remember exactly now. I should not have said "reverted" - I meant "changed/fixed" - I think I was building against some non-quite-ready commits when the issue happened.

All of my boards I've shipped are flashed with UF2 bootloaders built again my PR and are all working great!

hathach, 2022-02-07:
if it is not reproducible, it is not an issue anymore. Thanks for the udate