adafruit/tinyuf2 #188: ESP32-C3

jackqfzhang, 2022-01-24:
when can you support ESP32-C3?

skieast, 2022-01-24:
The USB peripheral on the C3 is not usable for TinyUF2. From Espressif
"Note that, in contrast with the USB OTG peripheral in some Espressif chips, the USB Serial/JTAG Controller is a fixed function device, implemented entirely in hardware. This means it cannot be reconfigured to perform any function other than to provide a serial channel and JTAG debugging functionality."

ladyada, 2022-01-24:
yes it is not possible - its not a true USB native device.

closed 2022-01-24 by ladyada

jackqfzhang, 2022-01-24:
it will crash to  run any code   after flashing CircuitPython 7.1.1 on my esp32C3 , I assumed  the cause is  no uf2 driver,  there is no way to run python code on esp32-c3?

tannewt, 2022-01-24:
uf2 is separate from CircuitPython. You'll want the `Absolute Newest` version of CircuitPython because we've fixed some crashes last. Please note that it's for advanced users only right now because the CIRCUITPY drive cannot work without native USB. In the long term we want to support the BLE workflow but that isn't currently working.

jackqfzhang, 2022-01-24:
thanks for your info.