adafruit/tinyuf2 #198: ESP32 S3 C1 N8R2 stays in boot mode after CircuitPython install

Craigxyzzy, 2022-03-31:
### Operating System

MacOS

### INFO_UF2.TXT

TinyUF2 Bootloader 0.8.0 - tinyusb (0.12.0-203-ga4cfd1c6)
Model: Espressif ESP32S3 DevKitC 1
Board-ID: ESP32S3-DevKitC-v1.0
Date: Jan  5 2022

### What happened ?

After copy of cp 72x onto S3DKC1BOOT and reboot, device remains in boot mode.

see https://forums.adafruit.com/viewtopic.php?f=60&t=189819

### How to reproduce ?

see https://forums.adafruit.com/viewtopic.php?f=60&t=189819

### Debug Log

none

hathach, 2022-04-01:
Make sure you 
- double check the N8R2 on the actual module, mismatched PSRAM configure could cause reset issue
- I have try https://circuitpython.org/board/espressif_esp32s3_devkitc_1_n8r2/ on my N8R2 and it flashed and could run circuitpython just fine  

One of the difference is, I use the latest tinyuf2 instead of release version, maybe you should try to see if that fix the issue:
- click on Artifacts to download https://github.com/adafruit/tinyuf2/actions/runs/2056501200
  [image]
- extract to get the combined.bin
- flash with `esptool.py --chip esp32s3 write_flash 0x0 combined.bin` (esptool.py should be v3.3 installed via pip isntall esptool)

Craigxyzzy, 2022-04-01:
Thank you!  That fixed it.  Now working with cp 724 (public release)

closed 2022-04-01 by Craigxyzzy

hathach, 2022-04-01:
> Thank you! That fixed it. Now working with cp 724 (public release)

can you confirm that the latest combined.bin fixed the issue, or just using the cp724 fix the issue ?

Craigxyzzy, 2022-04-01:
see https://forums.adafruit.com/viewtopic.php?f=60&t=189819
combined.bin fixed the issue, not cp724
FYI - initially tried drag copy update-tinyuf2.uf2, which did not resolve issue
did not try online serial tool
used esptool v3.3 to install combined.bin
first try failed - had not switched from USB to UART port
second try succeeded - connected to UART port
no plan to regression test cp720 .. 724

Any idea what changed in combined.bin

hathach, 2022-04-01:
thanks for confirmation, there is quite a lots of changed, I will make an release as 0.9.0 soon enough.

Craigxyzzy, 2022-04-01:
You are welcome!

Can you update INDEX.HTM for the s3 devkit c1 boards:

OLD user-guide-devkitm-1.html
NEW user-guide-devkitc-1.html

Thanks!

hathach, 2022-04-02:
thanks, I will make an update. Actually, could you do another test, so far we have
- web_esptool + 0.8.0 --> issue
- esptool.py + github combined --> work well

would you try to test with following combination just to rule out that web_epstool is the one that cause the issue (since S3 is added recently)
- esptool.py + 0.8.0

Craigxyzzy, 2022-04-04:
via USB port:
drag file onto boot drive
update-tinyuf2-080.uf2 <- SUCCESS
update-tinyuf2-9a414ac.uf2 <- SUCCESS

web-serial combined-080.bin <- FAIL
web-serial combined-9a414ac.bin <- SUCCESS

via UART port:
esptool combined-080.bin <- FAIL
esptool combined-9a414ac.bin <- SUCCESS

hathach, 2022-04-04:
thank you for detail testing, 0.9.0 is released with url fixed for devkitc1 https://github.com/adafruit/tinyuf2/releases/tag/0.9.0