adafruit/tinyuf2 #203: ESP32-S2 enter bootloader with magic number instead of RC pin

jonspieg, 2022-05-03:
I have a board I've designed that does not have the RC components and I want it to be able to enter bootloader by double tap and programmatically from within the firmware. this both could be implemented more easily using the magic number method. 

Is there a reason why the ESP32S2 boards implement it wit the RC instead of magic munber?

hathach, 2022-05-04:
Have you tried implementing the magic number approach ?

closed 2022-05-04 by hathach

ladyada, 2022-05-04:
esp32 clears out memory between resets, unlike other chips. RC 1 bit memory is required until someone invents something better

jonspieg, 2022-05-05:
Thanks, now I understand the rationale behind the RC pin.

I have a couple of questions to get a more complete picture.
1. after looking at the code for the grandcentral bootloader it saves the magic number in RAM (in the last address). Isn't RAM always cleared between resets by definition. or resetting using the reset pin doesn't actually cut power to the RAM? The ESP reset pin does cut power to the RAM?
2. Why not simply store it in NVM? Is it too slow? Will it decrease the life span of the flash chip? all of the above?

Now that I think of it, if I just want the ability to enter the bootloader from the firmware itself, using NVM magic number is probably the way to go, since it will be used quite rarely. Is there an obvious reason why this is not a good idea?

Thanks again!

hathach, 2022-05-06:
> 1. after looking at the code for the grandcentral bootloader it saves the magic number in RAM (in the last address). Isn't RAM always cleared between resets by definition. or resetting using the reset pin doesn't actually cut power to the RAM? The ESP reset pin does cut power to the RAM?

not possible since whole power for the chip is cut off

> 2. Why not simply store it in NVM? Is it too slow? Will it decrease the life span of the flash chip? all of the above?

only answer is giving it a try, if it works great, then please consider to submit an PR. It could be an option and help others.