adafruit/tinyuf2 #208: Possible incorrect BOARD_BOOT_LENGTH  being stuff into boot_data struct

ehughes, 2022-05-11:
### Operating System

Windows 10

### INFO_UF2.TXT

n/a

### What happened ?

**Summary:**

I believe the value use for BOARD_BOOT_LENGTH is incorrect.   In ports/mimxrt10xx/boards.h    

```
#define BOARD_BOOT_START        (((uint32_t) _ivt_origin) - 0x1000)

#define BOARD_BOOT_LENGTH       (0x8800) 
```

0x8800 is the value used for _text_length in ports/mimxrt10xx/linker/memory.ld

Since the ROM boot loader is copying down BOARD_BOOT_LENGTH from (((uint32_t) _ivt_origin) - 0x1000),   I believe  some code in the .text section will not get copied if you the image starts to get large. (only 0x6800 bytes)

I **think** the correct fix is something like

#define BOARD_BOOT_LENGTH   (0x8800 + 0x2000)

but it would probably be nice to have the values driven from the the values used in the .ld files.

**Backstory:**

The vanilla build of tinyuf2 for the MIMXRT1021 would yield a text section of 24792 bytes.  (0x60d8 bytes) which ran fine.

I noticed that as I added in the LOG=1 and a few extra init routines for an LCD on a board,  I would get goofy behavior on cold power up that would **not** occur when loading directly into itcm.    The CPU would end up hard faulting.   My modified program that enabled logging and some IO was getting close to the 0x8800 text section size (32560 bytes)

Changing BOARD_BOOT_LENGTH  to something like (0x8800 + 0x2000)  fixed the issue.    If I understand the linker .ld files correctly for the RT1021,   the fcfb and ivt will use up an extra 0x2000 bytes that need to get copied down.

If you could please take a looking and verify my thinking/analysis.   I don't think anyone would notice this w/ vanilla builds but it come pop up on another RT board port if more code was added.

### How to reproduce ?

n/a

### Debug Log

n/a

hathach, 2022-06-20:
just checking, did we manage to fix it previously. I kind of forgot.

ehughes, 2022-06-20:
It has not been fixed.   I am pretty sure what I suggested is the fix.  BOARD_BOOT_LENGTH needs to be whatever _text_length (from the memory.ld file) + 0x2000 to work properly.   

I

hathach, 2022-06-20:
if so, would you make an PR. I forgot most of the setup with imxrt, but I think other may hep to review the PR.

ehughes, 2022-06-20:
OK. Done.

FYI,  My port has been working good.     It is now integrated in the latest model of some fireworks equipment:

https://fireone.com/

The new 2022 hardware has tinyuf2 built in 😊

hathach, 2022-07-14:
hi @ehughes, sorry for the super late response, I got into too many bugs lately. And somehow this fell off my radar

> The new 2022 hardware has tinyuf2 built in blush

Congrats, I am happy to here that tinyuf2 is stable enough to be deployed on your new hardware.

hathach, 2022-07-15:
fixed by #220

closed 2022-07-15 by hathach