adafruit/tinyuf2 #211: ESP32-S2: Entering bootloader from FW using RTC memory magic number

jonspieg, 2022-05-19:
### Operating System

Windows 10

### INFO_UF2.TXT

not relevant

### What happened ?

I have a custom board that implements the esp32-s2 solo module.
I've attempted resetting the board into bootloader mode using the following snippet from the readme:
`#include "esp_private/system_internal.h"
void reboot_to_uf2(void)
{
  // Check out esp_reset_reason_t for other Espressif pre-defined values
  enum { APP_REQUEST_UF2_RESET_HINT = 0x11F2 };

  // call esp_reset_reason() is required for idf.py to properly links esp_reset_reason_set_hint()
  (void) esp_reset_reason();
  esp_reset_reason_set_hint(APP_REQUEST_UF2_RESET_HINT);
  esp_restart();
}`

This worked perfectly when compiled by the Arduino IDE.
However, when I used platformio with the same code, it just went into reset loop.

There was a very slight difference in the boot messages the chip spits out from uart0:
from arduino ide it was "rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)"
from platformio it was "rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)"

this might be related, or it might be irrelevant.

### How to reproduce ?

flash example code using platformio on an esp32-s2 dev board

hathach, 2022-05-19:
Closed since clearly not a bug of this repo. Check your flashing scheme, layout, linking etc of platformio and arduino

closed 2022-05-19 by hathach

hathach, 2022-05-24:
it can be anything, I don't use PIO to give an advise. For further question, please post on Adafruit support forum. Github is mostly for code related issue.