adafruit/tinyuf2 #216: [ESP32] Combine a default application with TinyUF2 binary

203Null, 2022-06-12:
I tried to put my application at the ota_0 partition but TinyUF2 would not boot it until an upload has occurred.
Upon checking the flash content, apparently, TinyUF2 has written something into nvs and otadata partition. 

Is there any way to create a TinyUF2 binary that contains a precompiled binary file?

hathach, 2022-06-15:
can you elaborate what you do exactly ?
1. what is your testing board, and what is its pre-flashing layout
2. which binary is flashed to which parittion, and your flashing command
3. what is the layout after flashing command.

203Null, 2022-06-16:
> can you elaborate what you do exactly ?
> 
> 1. what is your testing board, and what is its pre-flashing layout
> 2. which binary is flashed to which parittion, and your flashing command
> 3. what is the layout after flashing command.

1. I'm using my own board but we can treat it as esp32-s3-devkitm-1.
2. I'm flashing them via ESP32's flash download tool for testing. I flashed the bootloader manually with the following configuration generate by the build.
> --flash_mode dio --flash_freq 80m --flash_size 8MB
> 0x0 bootloader/bootloader.bin
> 0x410000 tinyuf2.bin
> 0x8000 partition_table/partition-table.bin
> 0xe000 ota_data_initial.bin

Then, I flash my firmware directly into the ota_0 partition at 0x10000.

3.  It will be the same as the partitions-8MB.csv.
> ESP-IDF Partition Table
> Name,   Type, SubType, Offset,  Size, Flags
> bootloader.bin,,          0x1000, 32K
> partition table,,         0x8000, 4K
> nvs,      data, nvs,      0x9000,  20K,
> otadata,  data, ota,      0xe000,  8K,
> ota_0,    0,    ota_0,   0x10000,  2048K,
> ota_1,    0,    ota_1,  0x210000,  2048K,
> uf2,      app,  factory,0x410000,  256K,
> ffat,     data, fat,    0x450000,  3776K,

hathach, 2022-06-16:
> 0x0 bootloader/bootloader.bin

this is incorrect offset for bootloader, should be 0x1000. Make sure you flash it with 

```
make BOARD=your-board flash
```

or its equivalent idf.py flash. There is flash_args file in the build folder that you can use.

203Null, 2022-06-16:
> > 0x0 bootloader/bootloader.bin
> 
> this is incorrect offset for bootloader, should be 0x1000. Make sure you flash it with
> 
> ```
> make BOARD=your-board flash
> ```
> 
> or its equivalent idf.py flash. There is flash_args file in the build folder that you can use.

Not for ESP32-S3, its bootloader partition does indeed starts at 0x0.

203Null, 2022-06-16:
> > Then, I flash my firmware directly into the ota_0 partition at 0x10000.
> 
> It won't be enough, you also need to set otadata (parition) to indicate the ota0 is valid partition. please check the esp doc for detail. TinyUF2 does also set this at the end of DFU if you are wondering on the code.

Thank you, that was what I was wondering. I'm not sure did TinyUF2 write something custom into otadata/NVS or is that something standard.

hathach, 2022-06-17:
that is not custom info, that is how otadata is intended to be used. Basically indicate which partition ota0, ota1 has a valid firmware and which one is active.