adafruit/tinyuf2 #218: cannot disable PIN_DOUBLE_RESET_RC feature

jonspieg, 2022-06-15:
### Operating System

Windows 10

### INFO_UF2.TXT

TinyUF2 Bootloader 0.9.0-5-g71a22e5-dirty - esp-idf (v4.4-dev-3608-gbbe2a1bf34) tinyusb (0.12.0-203-ga4cfd1c6)
Model: Adafruit Metro ESP32-S2
Board-ID: ESP32S2-Metro-revC
Date: Jun 15 2022

### What happened ?

I'm working on a custom board with the ESP32-S2 and I've used the same neopixel pin as in the adafruit metro esp32-s2 board. I've used the bootloader definition for the metro board, however I've commented out the line:
#define PIN_DOUBLE_RESET_RC   38
because I don't need this functionality (and I don't have a dedicated RC circuit on this pin).

However, I don't think commenting this out has any effect and this causes troubles for my specific board, which has some unrelated circuitry and behavior over on pin 38. The effect is that whenever I rest the board (single reset) it enters the bootloader, which is bad.

I also tried changing the pin number to a one not routed on my board, but still pin 38 is the one that affects the bootloader.

I changed my code to not interact with pi 38 at all, and now it doesn't go into bootloader on reset.

Is there anything special about pin 38?

### How to reproduce ?

this simplified sketch enters the bootloader (even though PIN_DOUBLE_RESET_RC is not defined)

void setup() {
    pinMode(38,OUTPUT);
    digitalWrite(38, HIGH);
    delay(1000);
    esp_restart();
}

note that some RC circuitry on pin 38 is probably needed for this to reproduce. in my case I guess the IC I have connected to this pin provides enough capacitance to retain the pin state during a power cycle.

hathach, 2022-06-16:
commenting out the macro will remove the respecting code, there is no question for that 
https://github.com/adafruit/tinyuf2/blob/7b3933c925f398ce3d72530372252f1adcabd2a6/ports/espressif/components/bootloader/subproject/main/bootloader_start.c#L213

make sure you re-flash the boot2 stage bootloader, also please provide your uart output.

jonspieg, 2022-06-16:
I have a great deal of confidence that I'm indeed flashing the correct bootloader because I've also tried changing the UF2_VOLUME_LABEL and it does take effect. 

I'm flashing the newly compiled bootloader by simply plugging in the new tinyuf2.bin into the existing arduino flashing command. should I change anything else besides that?

I will try to poke around the serial output and post it here.

jonspieg, 2022-06-19:
I tried it now with a proper metro esp32-s2 board - I commented out the PIN_DOUBLE_RESET_RC and changed the volume label to make sure I'm flashing the correct bootloader.
The volume label does change, but I still have the double tap to enter bootloader capability. I have no explanation for that. 
could anyone else try to reproduce it?

I might be flashing my custom bootloader incorrectly. What I did is just to copy the command line from the Arduino ide and replaced the tinyuf2.bin on address 0x2d0000. Is there anything else I should have done in order to properly flash the custom bootloader?

Thanks

jonspieg, 2022-06-20:
Just realized there's a separation between the tinyuf2 firmware and the actual 2nd stage bootloader which is a different bin file at address 0x1000. 
When replacing both files in the command line everything works as expected.

thanks

closed 2022-06-20 by jonspieg