adafruit/tinyuf2 #219: feather_stm32f405_express not running blinky uf2

maca134, 2022-06-17:
### Operating System

Windows 10

### INFO_UF2.TXT

```
TinyUF2 Bootloader 0.9.4 - st/cmsis_device_f4 (v2.6.4-1-g2615e86) st/stm32f4xx_hal_driver (v1.7.9) tinyusb (0.12.0-203-ga4cfd1c69)
Model: Adafruit Feather STM32F405 Express
Board-ID: STM32F405-Feather-revB
Date: May 30 2022
```

### What happened ?

TinyUF2 doesnt seem to be running programs copied over to the MSC.

### How to reproduce ?

1. Compile blink example for feather_stm32f405_express
2. Convert bin to uf2 `uf2conv -c -b 0x08010000 -f 0x57755a57 -o...`
3. Copy uf2 to MSC device
4. Board resets back into tinyuf2 bootloader (neopixel is green + MSC device mounts)

### Screenshots

Bootloader is there
[image]

Blink program seems to start at 0x08010000
[image]

hathach, 2022-06-17:
can you attach both the bin and uf2 file here

hathach, 2022-06-20:
I should have asked this before, but can you also share your simplest test project as github repo viewable and commentable for review. Make sure you got the linker correctly setup with flash layout for application (staring at correct address). It is probably better to convert with hex file than bin file.

I will find one example that works as well for your reference when I got time to pull out the board to test with.

hathach, 2022-06-21:
I think the issue is probably you didn't have correct linker script for your firmware application. The starting address must be `0x08010000` since vector table along with function call may or probably use absolute address. Can you double check that ?

hathach, 2022-06-23:
also if you are using cmsis f4 from ST, make sure you update to the latest, old version will incorrectly set the vector table. I have update readme for this note
https://github.com/adafruit/tinyuf2/blob/master/ports/stm32f4/README.md

vibr77, 2024-10-20:
I have the same issue with a STM32F411,

hathach, 2025-02-11:
it is probably due to the VTOR is set to 0x0 by system_stm32xxx.c try to add this line to your main() https://github.com/adafruit/tinyuf2/pull/419/files#diff-19aa9dc949d3bb8855bb23418f05fc68134dcbc6a74aa6a63a4f311d8a3adc45R67 . I will revise and add blinky app for testing for this port later.