adafruit/tinyuf2 #227: ESP32-SX Have alternate partition scheme with no OTA support

UnexpectedMaker, 2022-06-25:
Though OTA is cool, I'd hazard to guess that fewer CircuitPython users would use OTA in general, and even be confident enough to guess hardly any use it all.

I'd like to see, not only a second UF2 option for ESP32-SX boards that has no OTA partitions laid out at all, and just a much bigger user file system partition, but also have that be the new default partition layout/binary and have OTA as a second option for those that want to use OTA.

I've spoken a few times with @tannewt about this, and it has to happen at the UF2 bootloader side, as that is what enforces the partition scheme, so we'd need this integrated as alternate builds for ESP32 (and any other platforms that could/might) use this in the future.

I am constantly getting asked questions like... 
[image]

or folks just letting me know their board they just received is faulty because it's only got 2MB of flash on it instead of 4MB or 8MB etc like advertised.

I don't really mind how this is implemented - but I think it's really important to give users more flash if it's available on the chip, and the choice to use it or choose OTA instead if that suits their needs.

@hathach Thoughts?

Neradoc, 2023-02-23:
Note that a solution is now released in Circuitpython 8. This PR: https://github.com/adafruit/circuitpython/pull/7000 thanks to MicroDev, lets you reformat the CIRCUITPY drive so that it extends into the OAT1 partition.
https://docs.circuitpython.org/en/latest/shared-bindings/storage/index.html#storage.erase_filesystem

- An empty flash is formatted in extended mode by default.
- It disables the dualbank module (it's still there but raises an exception).
- Reformatting an existing drive preserves the [non] extended nature of it.

An existing drive can be extended like this (save your data first of course).
```py
import storage
storage.erase_filesystem(extend=True)
```

hopkapi, 2023-03-06:
Just to be clear, it's actually "extended", rather than "extend":

```
import storage
storage.erase_filesystem(extend=True)
```

PR #336 (2023-07-19): Implement espressif 1 ota partition
## Description of Change

- Add partitions with 1 factory (tinyuf2) + 1 ota app. Fix #227
- Change adafruit_camera_s3 to use 1 ota app partition
- Added Flash size: entry to INFO_UF2.TXT to make it easier to spot the different in partitions
```
TinyUF2 Bootloader 0.15.0-23-g598d624 - esp-idf (v4.4-dev-3608-gbbe2a1bf34) tinyusb (0.15.0-566-gf1e006d09)
Model: Adafruit Camera
Board-ID: ESP32S3-Camera-revC
Date: Jul 19 2023
Flash Size: 0x002C0000 bytes
```
- Fixed ghostfat selftest check with new "Flash Size" item, which actually takes more time to add the parit [... cut]
closed 2023-07-19 by hathach