adafruit/tinyuf2 #231: Flash issue with RT1024 EVK

hathach, 2022-07-15:
### Operating System

Linux

### INFO_UF2.TXT

TinyUF2 Bootloader 0.10.2-21-g71a89e4-dirty - nxp/mcux-sdk (MCUX_2.10.0-25-ga242602a) tinyusb (0.12.0-203-ga4cfd1c6)
Model: NXP RT1024 EVK
Board-ID: MIMXRT1024-EVK-revA
Date: Jul 15 2022

### What happened ?

- `flash-sdp` work, bootloader seems to run well on SRAM, and seems to write to flash
- however when reset, the booROM run again instead of the tinyuf2 even though the boot switch is set to normal `10`

This seems to indicate there is some misconfigured in the fcfb that cause bootROM to run instead of loading tinuf2. I have tried to alter several flash_config but does not seem to work. 

NOTE: rt1020 evt work well with an 8MB external flash, the rt1024 with built-in qspi flash probably need a bit more testing.

### How to reproduce ?

1. power rt1024 with BOOTSW = `10`
2. `make BOARD=imxrt1024_evk LOG=1 flash-sdp`
3. TinyUF2 will show up as RT1024BOOT
4. press reset
5. the bootROM will run instead of tinyuf2
```
[ 7621.815570] usb 3-2: New USB device found, idVendor=1fc9, idProduct=0130, bcdDevice= 1.01
[ 7621.815576] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7621.815578] usb 3-2: Product: SE Blank RT Family 
[ 7621.815580] usb 3-2: Manufacturer: NXP      SemiConductors Inc 
```

### Debug Log

n/a

hathach, 2023-03-31:
There is a recent pr fixed by other users to tinyusb that fixes the 1024 issue. I will pull out and check this later next week when I get back from vacation.

gsteiert, 2023-04-01:
TinyUF2 only writes itself to flash under certain conditions (from board_flash.c):
```
  // TinyUF2 will copy its image to flash if one of conditions meets:
  // - Boot Mode is '01' i.e Serial Download Mode (BootRom)
  // - Flash FCFB is invalid e.g blank flash
  // - Flash contents does not match running tinyuf2 in SRAM (depending on COMPARE_AND_WRITE_TINYUF2)
```

Here are four ways to force it to load:

1. Change BOOTSW to 01
2. Erase flash before loading to clear FCFB
4. Drag-n-drop the TinyUF2.uf2 self update image after loading
3. Change COMPARE_AND_WRITE_TINYUF2 define to 1

tannewt, 2023-04-03:
I think I got TinyUF2 writing to flash ok.

What I don't understand is that the RAM version is linked to run from the address range of RAM. How does that work when loading/running from flash?

tannewt, 2023-04-03:
@gsteiert Any pointers to where this is documented? How does the RAM address need to relate to the flash address? Does the start of the image start with the boot data?

gsteiert, 2023-04-03:
See the program image section of the reference manual (9.7 in RT1010)
The i.MX RT Flashloader Use Case app note also has some additional information:
https://www.nxp.com/docs/en/nxp/application-notes/AN12238.pdf