adafruit/tinyuf2 #242: STM32F401CCU6

Bl00d0und4, 2022-09-04:
### Operating System

Windows 11

### INFO_UF2.TXT

TinyUF2 Bootloader 0.10.2 - st/cmsis_device_f4 (v2.6.8) st/stm32f4xx_hal_driver (v1.8.0-8-g1d99564) tinyusb (0.12.0-203-ga4cfd1c69)
Model: STM32 STM32F401CxUx
Board-ID: BlackPill
Date: Jun 24 2022

### What happened ?

I successfully flashed the controller with tinyuf2. The USB drive appears and I can copy the UF2 file to the controller. Then the controller restarts but the program is not loaded.

### How to reproduce ?

I convert a .bin with the following commands:
uf2conv.py -c -b 0x08010000 -f STM32F4 beta_1.01.ino.bin

hathach, 2022-09-08:
simply converting binary with uf2conv won't work, you will need to change the linker to start with 0x8010000 https://github.com/adafruit/tinyuf2/issues/75#issuecomment-889666036

closed 2022-09-08 by hathach

Bl00d0und4, 2022-09-08:
i dont know how :-(

hathach, 2022-09-08:
for how-to, you could use discussion, post your detail setup and link to your github project. Others may help you out.