adafruit/tinyuf2 #243: Cannot disable write protection on STM32F401

myst729, 2022-09-08:
## Is your feature request related to a problem? Please describe.

Recently I flashed 0.10.2 to my STM32F401 breakout board. When I try to erase it and flash another firmware, it fails.

By checking release notes I found write protection was introduced in an early version. Following KarlK90's guide, I build a [tinyuf2-unlocker](https://github.com/KarlK90/tinyuf2-unlocker/issues/1) uf2 file for STM32F401. When I drag the uf2 to the blackpill drive, it ejects. Then I flash my new bin file with dfu-util, it says successful (100% done). However, after a reboot the blackpill drive appears again, it doesn't unlock. Later I also tried STM32Cube and CMSIS-DAP with no luck.

I'm not familiar with rust, so I don't know whether I did anything wrong in building tinyuf2-unlocker. But it looks fine to me, no errors occur in the whole process.

According to [hathach's reply](https://github.com/adafruit/tinyuf2/pull/189#issuecomment-1163257209), my CMSIS-DAP might need to upgrade firmware. This has to be done from the host over another downloader, but unfortunately I don't have an extra **-link.

## Describe the solution you'd like

A working unlock tool for STM32F401 that doesn't require a downloader would be great.

## Describe alternatives you've considered

Maybe make the protection an opt-in in future releases?

## Additional context

Flashing with CMSIS-DAP:

```
$ pyocd flash -t stm32f401cdux dg_f401_default.bin
0001939:INFO:load_cmd:Loading /Users/leo/Downloads/dg_f401_default.bin
[                                                  ]   0%0007318:CRITICAL:__main__:flash erase sector failure (address 0x08000000; result code 0x1)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/pyocd/__main__.py", line 150, in run
    status = cmd.invoke()
  File "/usr/local/lib/python3.9/site-packages/pyocd/subcommands/load_cmd.py", line 117, in invoke
    programmer.program(filename,
  File "/usr/local/lib/python3.9/site-packages/pyocd/flash/file_programmer.py", line 171, in program
    self._loader.commit()
  File "/usr/local/lib/python3.9/site-packages/pyocd/flash/loader.py", line 289, in commit
    perf = builder.program(chip_erase=chipErase,
  File "/usr/local/lib/python3.9/site-packages/pyocd/flash/builder.py", line 509, in program
    flash_operation = self._sector_erase_program_double_buffer(progress_cb)
  File "/usr/local/lib/python3.9/site-packages/pyocd/flash/builder.py", line 911, in _sector_erase_program_double_buffer
    self.flash.erase_sector(sector.addr)
  File "/usr/local/lib/python3.9/site-packages/pyocd/flash/flash.py", line 374, in erase_sector
    raise FlashEraseFailure('flash erase sector failure', address=address, result_code=result)
pyocd.core.exceptions.FlashEraseFailure: flash erase sector failure (address 0x08000000; result code 0x1)
```

hathach, 2022-09-08:
protect bootloader is already optional, you only need to define `TINYUF2_PROTECT_BOOTLOADER=0` in your board if you don't want it. https://github.com/adafruit/tinyuf2/blob/master/ports/stm32f4/boards.h#L47

developing bootloader always need an programmer. Though self-update is possible, you can compile tinyuf2 without protect bootloader and do the upgrade.

closed 2022-09-08 by hathach

myst729, 2022-09-08:
The problem is I have already flashed a firmware that enables 
write protection from https://github.com/adafruit/tinyuf2/releases/download/0.10.2/tinyuf2-stm32f401_blackpill-0.10.2.zip, how can I erase it.