adafruit/tinyuf2 #269: Support for ESP32-S3-DevKitC-1-N16R8V and ESP32-S3-DevKitC-1-N32R8V

rgrizzell, 2022-12-22:
### Operating System

Linux

### INFO_UF2.TXT

TinyUF2 Bootloader 0.11.0-dirty - esp-idf (v4.4-dev-3608-gbbe2a1bf34) tinyusb (0.12.0-203-ga4cfd1c69)
Model: Espressif ESP32S3 DevKitC 1
Board-ID: ESP32S3-DevKitC-v1.0
Date: Dec 21 2022

### What happened ?

Espressif recently release a revision of the ESP32-S3 using the **WROOM-2** module dubbed: _ESP32-S3-DevKitC v1.1_. They include the following models:
**ESP32-S3-DevKitC1-N16R8V (16MB Flash, 8MB PSRAM)**
**ESP32-S3-DevKitC1-N32R8V (32MB Flash, 8MB PSRAM)**
(As far as I can tell, the "V" is for flash that runs at 1.8V instead of 3.3V).

The current pre-compiled builds for the ESP32-S3-DevKitC-1 do not support devices with more than 8MB of flash. Additionally, the currently linked revision of ESP-IDF does not include a version of esptool.py that supports >16MB of Flash.

Outward appearances lead me to believe that it's necessary to [upgrade ESP-IDF to v5.0](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/index.html).

### How to reproduce ?

1. Erase: `esptool.py --chip esp32s3 -p /dev/ttyACM0 erase_flash`
2. Build: `make BOARD=espressif_esp32s3_devkitc_1 all`
3. Flash: `make BOARD=espressif_esp32s3_devkitc_1 flash`
3a. Workaround: Add `CONFIG_FLASHMODE_DIO=y` to `ports/espressif/boards/espressif_esp32s3_devkitc_1/sdkconfig`

### Debug Log

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x1508
load:0x403b6000,len:0xb44
load:0x403ba000,len:0x2e1c
entry 0x403b62e8
I (32) boot: ESP-IDF v4.4-dev-3608-gbbe2a1bf34 2nd stage bootloader
I (32) boot: compile time 10:53:06
I (32) boot: chip revision: 0
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : SLOW READ
I (45) boot.esp32s3: SPI Flash Size : 8MB
I (50) boot: Enabling RNG early entropy source...
W (55) bootloader_random: RNG for ESP32-S3 not currently supported
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (80) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (88) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (95) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (103) boot:  4 uf2              factory app      00 00 00410000 00040000
I (110) boot:  5 ffat             Unknown data     01 81 00450000 003b0000
I (118) boot: End of partition table
I (122) boot: Defaulting to factory image
I (127) esp_image: segment 0: paddr=00410020 vaddr=3c020020 size=03c9ch ( 15516) map
I (139) esp_image: segment 1: paddr=00413cc4 vaddr=3fc8fc60 size=01b78h (  7032) load
I (146) esp_image: segment 2: paddr=00415844 vaddr=40374000 size=0a7d4h ( 42964) load
I (164) esp_image: segment 3: paddr=00420020 vaddr=42000020 size=12f28h ( 77608) map
I (183) esp_image: segment 4: paddr=00432f50 vaddr=4037e7d4 size=01488h (  5256) load
I (185) esp_image: segment 5: paddr=004343e0 vaddr=50000000 size=00010h (    16) load
I (193) boot: Loaded app from partition at offset 0x410000
I (195) boot: Disabling RNG early entropy source...
W (200) bootloader_random: RNG for ESP32-S3 not currently supported
I (219) cpu_start: Pro cpu up.
I (219) cpu_start: Starting app cpu, entry point is 0x40374ce0
I (0) cpu_start: App cpu up.
W (225) rtcinit: calibration efuse version does not match, set default version: 0

I (242) cpu_start: Pro cpu start user code
I (242) cpu_start: cpu freq: 160000000
I (242) cpu_start: Application information:
I (244) cpu_start: Project name:     tinyuf2
I (249) cpu_start: App version:      0.11.0-dirty
I (255) cpu_start: Compile time:     Dec 21 2022 10:53:03
I (261) cpu_start: ELF file SHA256:  5fa21b798a3c1935...
I (267) cpu_start: ESP-IDF:          v4.4-dev-3608-gbbe2a1bf34
I (274) heap_init: Initializing. RAM available for dynamic allocation:
I (281) heap_init: At 3FCA4EB0 len 0003B150 (236 KiB): D/IRAM
I (287) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (294) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (300) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (311) spi_flash: detected chip: mxic
W (311) spi_flash: Detected flash size > 16 MB, but access beyond 16 MB is not supported for this flash model yet.
I (321) spi_flash: flash io: dio
W (325) spi_flash: Detected size(32768k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (339) sleep: Configure to isolate all GPIO pins in sleep state
I (345) sleep: Enable automatic switching of GPIO sleep configuration
I (352) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (61328) esp_image: image at 0x10000 has invalid magic byte (nothing flashed here?)
```

rgrizzell, 2022-12-22:
This comment goes into more detail about the esptool.py flashing limitation.
https://github.com/adafruit/tinyuf2/issues/268#issuecomment-1356862612

In this comment I attempted a separate board definition.
https://github.com/adafruit/tinyuf2/issues/268#issuecomment-1356889129

hathach, 2022-12-23:
I don't have the new wroom2 board for now, will try to get one later

skieast, 2023-01-06:
Esptool does not support a flash of more that 16MB (except for a specific Gigadevice chip) .  This support is expected to be added in the future.   This is commented on in the changelog for the dev version of esptool.

```
The octal flash support for 32 MB chips
(https://github.com/espressif/esptool/issues/795) will be added in a
follow-up commit.
```

https://github.com/espressif/esptool/commit/21e59148e73e384947327a0883a753685e03d143

hathach, 2023-11-28:
IDF is updated to 5.1, there is also an partition csv with 16MB. Please try it out again to see if that works.

rgrizzell, 2023-11-29:
Looks great! I can see the full 32MB of flash using the [tinyuf2-espressif_esp32s3_devkitc_1-0.18.1.zip](https://github.com/adafruit/tinyuf2/releases/download/0.18.1/tinyuf2-espressif_esp32s3_devkitc_1-0.18.1.zip) artifact.

```
$ df -h
Filesystem             Size  Used Avail Use% Mounted on
...
/dev/sdb                32M  4.1M   28M  13% /media/rgrizzell/S3DKC1BOOT
```

---

For posterity, I'm including the output from the flash operation. I couldn't figure out if ESP-IDF is still included in this repository, but I was able to setup v5.1 in a separate directory.

```
$ esptool.py --chip esp32s3 write_flash 0x0 combined.bin
esptool.py v4.7.dev3
Found 3 serial ports
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_1v8)
Crystal is 40MHz
MAC: 68:b6:b3:3d:7a:88
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0043cfff...
Compressed 4442768 bytes to 136070...
Wrote 4442768 bytes (136070 compressed) at 0x00000000 in 46.9 seconds (effective 758.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

**UART**
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3818,len:0x16dc
load:0x403c9700,len:0x4
load:0x403c9704,len:0x1054
load:0x403cc700,len:0x3090
entry 0x403c9a68
I (34) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (34) boot: compile time Nov 14 2023 11:29:51
I (34) boot: Multicore bootloader
I (37) boot: chip revision: v0.1
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : SLOW READ
I (51) boot.esp32s3: SPI Flash Size : 8MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (79) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (87) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (94) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (102) boot:  4 uf2              factory app      00 00 00410000 00040000
I (109) boot:  5 ffat             Unknown data     01 81 00450000 003b0000
I (117) boot: End of partition table
I (121) boot: Defaulting to factory image
I (126) boot: Reset Reason = 1
I (130) esp_image: segment 0: paddr=00410020 vaddr=3c020020 size=07bf8h ( 31736) map
I (146) esp_image: segment 1: paddr=00417c20 vaddr=3fc90500 size=021dch (  8668) load
I (149) esp_image: segment 2: paddr=00419e04 vaddr=40374000 size=06214h ( 25108) load
I (162) esp_image: segment 3: paddr=00420020 vaddr=42000020 size=16828h ( 92200) map
I (185) esp_image: segment 4: paddr=00436850 vaddr=4037a214 size=06214h ( 25108) load
I (198) boot: Loaded app from partition at offset 0x410000
I (198) boot: Disabling RNG early entropy source...
I (210) cpu_start: Multicore app
I (210) cpu_start: Pro cpu up.
I (215) cpu_start: Starting app cpu, entry point is 0x40374d0c
I (0) cpu_start: App cpu up.
I (229) cpu_start: Pro cpu start user code
I (229) cpu_start: cpu freq: 160000000 Hz
I (229) cpu_start: Application information:
I (232) cpu_start: Project name:     tinyuf2
I (237) cpu_start: App version:      0.18.1
I (242) cpu_start: Compile time:     Nov 14 2023 11:29:37
I (248) cpu_start: ELF file SHA256:  d5f8f831d3f5195e...
I (254) cpu_start: ESP-IDF:          v5.1.1
I (259) cpu_start: Min chip rev:     v0.0
I (264) cpu_start: Max chip rev:     v0.99 
I (269) cpu_start: Chip rev:         v0.1
I (273) heap_init: Initializing. RAM available for dynamic allocation:
I (281) heap_init: At 3FCA60E8 len 00043628 (269 KiB): DRAM
I (287) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (294) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (300) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
W (306) spi_flash: Octal flash chip is using but qio mode is selected, will automatically swich to Octal mode
I (317) spi_flash: detected chip: mxic (opi)
I (321) spi_flash: flash io: opi_str
W (326) spi_flash: Detected size(32768k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (339) sleep: Configure to isolate all GPIO pins in sleep state
I (346) sleep: Enable automatic switching of GPIO sleep configuration
I (353) app_start: Starting scheduler on CPU0
32(3 8) ap _start:sStaStint scheduPer on CPU1
I (368) main_task: Calling app_main()
                                     I (368) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
            I (378) main_task: Returned from app_main()
```

closed 2023-11-29 by rgrizzell

hathach, 2023-11-29:
> For posterity, I'm including the output from the flash operation. I couldn't figure out if ESP-IDF is still included in this repository, but I was able to setup v5.1 in a separate directory.

IDF is not included as submodule anymore, since we will now try to target release version. Previous specific commit is required since esp-idf is still under develop to add s2/s3 support.