adafruit/tinyuf2 #270: Support encryption payload in uf2 file

howard0su, 2023-01-01:
**Is your feature request related to a problem? Please describe.**
I would like to encrypt uf2 file with AES128/256 encryption method so that nobody is able to disassembly the file.

**Describe the solution you'd like**
Add code logic to support uf2 file with some extension flags which indicated the firmware is encrypted and need run decrypt before flash it.

mickeydarrenlau, 2023-01-16:
Then where to put the keys

ccrome, 2023-03-02:
Definitely something I'm interested in too.

for the i.MX RT family, there are two different encryption schemes -- BEE and OTFAD.  Unfortunately at the moment, I don't know much more about it ;-)

hathach, 2023-03-03:
Before discussing the implementation, I would suggest to bring this discussion to https://github.com/microsoft/uf2 for the author to make it into the specs first. E.g adding new flag for encrypted method/key etc..

lhartmann, 2025-05-21:
Doesn't UF2 simply serves as a raw/binary transfer esncapsulation?

Can´t you just transfer the encrypted copy of your firmware, and have your custom code decrypt before flashing.