adafruit/tinyuf2 #276: CI Deprecation Warnings

tekktrik, 2023-01-31:
### Operating System

Others

### INFO_UF2.TXT

N/A

### What happened ?

GitHub Actions is deprecating Node.js 12 and also changing how outputs are set in steps.

### How to reproduce ?

Run Actions.

tekktrik, 2023-01-31:
I can work on this but wanted an issue so I wouldn't forget.

hathach, 2023-01-31:
iirc we can fix this by update `actions/setup-python@v4` . @tekktrik have you tried any code changes just yet, if not, I could try to make an PR to see if that works out.

tekktrik, 2023-01-31:
I've updated versions for standard action steps like `actions/setup-python` and 'actions/checkout` in the CircuitPython libraries without issue.  Same thing for how the outputs of steps are set, no issues there either.

PR #281 (2023-02-07): Update CI actions to Node.js 16
## Checklist

*By completing this PR sufficiently, you help us to review this Pull Request quicker and also help improve the quality of Release Notes*

- [ X ] Please provide specific title of the PR describing the change
- [ X ] Please provide related links (*eg. Issue which will be closed by this Pull Request*)

*This checklist items that are not applicable to your PR can be deleted.*

-----------

## Description of Change

Updates CI files to stop using actions using Node.js 12.

Fixes #276
closed 2023-02-07 by hathach