adafruit/tinyuf2 #286: mimxrt1011, 400mhz support

ccrome, 2023-02-18:
### Operating System

Linux

### INFO_UF2.TXT

If i'm not mistaken, I believe that the default mimxrt1011 tinyuf2 starts at 500MHz.  It should probably start at 400Mhz or lower to ensure that it's compatible with all chip revisions.

I might be mistaken -- I used the mcu config tools to read the clock_config.c file, so it's quite possible that process didn't work right.

### What happened ?

I think the system is starting at 500Mhz, which is well outside the specs for the 400mhz part.

### How to reproduce ?

just putting something here because i have to.

tannewt, 2023-03-01:
The only version I see [here](https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/i-mx-rt-crossover-mcus/i-mx-rt1010-crossover-mcu-with-arm-cortex-m7-core:i.MX-RT1010) is 500mhz capable. Where is a 400mhz part documented?

ccrome, 2023-03-01:
> 

in the "[Buy/Paramettrics](https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/i-mx-rt-crossover-mcus/i-mx-rt1010-crossover-mcu-with-arm-cortex-m7-core:i.MX-RT1010?tab=Buy_Parametric_Tab#/)" tab.

[image]

hathach, 2023-03-02:
each board has its own clock_config.c/h and can be configured individually, yes ?

ccrome, 2023-03-02:
> each board has its own clock_config.c/h and can be configured individually, yes ?

Yes.  Now that I think about it, this really isn't a bug for the imxrt1010_evk.  It's that I was using the imxrt1010_evk as a placeholder for 'all i.mx RT 1011 based boards'.  

I guess it's just a policy choice:  do you want the clock speed set to the maximum clock speed available for that particular board, or the maximum clock speed supported by all members of the CPU family?  The former means when transitioning from an evk to a custom board, it's easy to miss the fact that the clock speed of the chip is different (exactly what happened to me), vs. the bootloader running at lower clock speed.

Actually, now that I think about it, does the bootloader need to run at such high speeds at all?  Perhaps it should just leave the PLL disabled and run everything at 24 MHz?

@gsteiert any opinion here?  I was wondering if we should do the minimum amount of clock setup necessary to make things go, so it's left for the app in as close to reset state as possible.  Rather than doing the full auto-generated BOARD_BootClockRUN call.

hathach, 2023-03-03:
Each board should configure its preferred clock individually with clock_config. I don't think we need to set any minimum or maximum clock at all. They should run with their optimum design (crystal/power etc...). If you accept this answer, we can close this issue.

gsteiert, 2023-03-03:
I agree it should be left to each board to decide. They may need different
settings for a different clock source or crystal frequency.  Also, some of
the settings may be required to run USB.

closed 2023-03-03 by ccrome