adafruit/tinyuf2 #287: iMXRT10xx Linker Issues

gsteiert, 2023-02-21:
### Operating System

Linux

### INFO_UF2.TXT

NA

### What happened ?

The .hex file generated by the linker script has the virtual memory addresses, not the load addresses, so it is not useful for loading the flash, with the Teensy loader for example.  This is my mistake, if I had made the linker scripts correctly, it should be able to create a valid .hex file with the correct flash addresses.  

### How to reproduce ?

These instructions are for the Teensy 4.x board
1. Attempt to load the generated .hex file with Teensy Loader. This will corrupt the flash so the board gets stuck in serial download mode.
2. Convert the .bin to .hex with this command `arm-none-eabi-objcopy -I binary -O ihex --change-addresses 0x60000000 tinyuf2-teensy41.bin tinyuf2-teensy41-fix.hex`  This will generate a .hex file with the proper addresses.

hathach, 2023-02-23:
it is totally not your mistake nor anyone mistake. We just simply have not tested the hex file just since migrated bootloader to SRAM. And the sdphost works perfectly since then. Let me pull out my rt1011 since it has on-board daplink that I can flash the hex file with pyocd to test out your suggestion of adding `--change-address`

hathach, 2023-03-11:
fixed by #291

closed 2023-03-11 by hathach