adafruit/tinyuf2 #300: self-update doesn't reboot after update

ccrome, 2023-03-07:
### Operating System

Linux

### INFO_UF2.TXT

placeholder

### What happened ?

When copying `update-tinyuf2-imxrt1010_evk.uf2` to the board, it does not reboot, and it requires a manual reset.

**expected behavior**
I would expect that the board reboots, just like it does when jumping to a new application.   In its current incarnation, unattended updates are not easy, and would require a full power off and on of the entire system, which is not feasible for some embedded systems.  

### How to reproduce ?

```
cp update-tinyuf2-imxrt1010_evk.uf2 /media/caleb/RT1010BOOT
```

would expect a reboot into the new bootlaoder.

ccrome, 2023-03-07:
@gsteiert I think I see what's going on here:

This issue may be limited to the i.MX RT port, which treats the self update exactly the same as a regular update.  The regular update jumps to the application directly, rather than rebooting.   This means, when the self-update happens, the flash gets written, but the chip doesn't go through the boot process, and so the new version doesn't get started.

Is there a way to force-boot on self-update?

ccrome, 2023-03-07:
smh, okay, I'll understand this eventually.  It actually *does* start up the new firmware, however, it does not re-enumerate on the USB bus.  

My experience is this:

I have 2 i.MX RT1010 builds:
```
make BOARD=imxrt1010_evk self-update
make BOARD=custom self-update
```

The drive for the `imxrt1010_evk` is `/media/$USER/RT1010BOOT`.   The Drive for `custom` is `/media/$USER/CSTMBOOT`.

I initially program with the `tinyuf2-imxrt1010_evk.bin` file, so when I reboot, I have the RT1010BOOT drive.

Now, I run the firmware update with `cp _build/custom/update-tinyuf2-custom.uf2 /media/$USER/RT1010BOOT` and in syslog I get:

```
Mar  7 12:31:14 cc-i9 kernel: [ 1396.063303] usb 1-9.1.4: reset high-speed USB device number 44 using xhci_hcd                                                                                                     
```

and the drive `/media/$USER/RT1010BOOT` stays where it is.  It doesn't go away or anything.  I can copy the file again, and I get another syslog entry that looks exactly the same.  But strangely, the new firmware is running.   

Seems as though there is some need to do an explicit USB disconnect or something, or perhaps a SystemReset to make sure everything starts fresh?

hathach, 2023-03-08:
can you provide the full dmesg log when first attached with old firmware then copy new firmware.

ccrome, 2023-03-09:
Ah ha.  I hadn't changed the PID.  Found that if I don't change the PID, it doesn't recognize the new firmware right.    If I do change the PID, then it re-enumerates.

I created a quick dummy board [here](https://github.com/ccrome/tinyuf2/tree/illustrate-300-reboot) , which just has some strings and PID changed from the `imxrt1010_evk` board.  No other code changes.

```bash
make BOARD=imxrt1010_evk -j flash
```
dmesg output
```
[  +5.309085] usb 1-9.1.4: USB disconnect, device number 118
[  +0.368973] usb 1-9.1.4: new high-speed USB device number 119 using xhci_hcd
[  +0.100967] usb 1-9.1.4: New USB device found, idVendor=239a, idProduct=0077, bcdDevice= 1.00
[  +0.000014] usb 1-9.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000006] usb 1-9.1.4: Product: RT1010 EVK
[  +0.000005] usb 1-9.1.4: Manufacturer: NXP
[  +0.000004] usb 1-9.1.4: SerialNumber: DA081561D7091730B500005002004100
[  +0.002476] usb-storage 1-9.1.4:1.0: USB Mass Storage device detected
[  +0.000398] scsi host30: usb-storage 1-9.1.4:1.0
[  +0.001457] hid-generic 0003:239A:0077.00D1: hiddev1,hidraw1: USB HID v1.11 Device [NXP RT1010 EVK] on usb-0000:00:14.0-9.1.4/input1
[  +1.015395] scsi 30:0:0:0: Direct-Access     Adafruit UF2 Bootloader   1.0  PQ: 0 ANSI: 2
[  +0.000680] sd 30:0:0:0: Attached scsi generic sg7 type 0
[  +0.000666] sd 30:0:0:0: [sdh] 65801 512-byte logical blocks: (33.7 MB/32.1 MiB)
[  +0.000446] sd 30:0:0:0: [sdh] Write Protect is off
[  +0.000007] sd 30:0:0:0: [sdh] Mode Sense: 03 00 00 00
[  +0.000353] sd 30:0:0:0: [sdh] No Caching mode page found
[  +0.000004] sd 30:0:0:0: [sdh] Assuming drive cache: write through
[  +0.002055]  sdh:
[  +0.000116] sd 30:0:0:0: [sdh] Attached SCSI removable disk
```

**Without PID change**
```bash
make BOARD=custom self-update
cp _build/custom/update-tinyuf2-custom.uf2 /media/caleb/RT1010BOOT/
```

```
[Mar 8 21:15] usb 1-9.1.4: reset high-speed USB device number 121 using xhci_hcd
```

**With PID Change**
```bash
make BOARD=custom self-update
cp _build/custom/update-tinyuf2-custom.uf2 /media/caleb/RT1010BOOT/
```

```
[Mar 8 21:13] usb 1-9.1.4: reset high-speed USB device number 119 using xhci_hcd
[  +0.103771] usb 1-9.1.4: device firmware changed
[  +0.000161] usb 1-9.1.4: USB disconnect, device number 119
[  +0.015747] device offline error, dev sdh, sector 129 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[  +0.000018] Buffer I/O error on dev sdh, logical block 129, lost async page write
[  +0.000155] device offline error, dev sdh, sector 387 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[  +0.000018] Buffer I/O error on dev sdh, logical block 387, lost async page write
[  +0.000035] device offline error, dev sdh, sector 517 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[  +0.000009] Buffer I/O error on dev sdh, logical block 517, lost async page write
[  +0.107650] FAT-fs (sdh): unable to read boot sector to mark fs as dirty
[  +0.007960] usb 1-9.1.4: new high-speed USB device number 120 using xhci_hcd
[  +0.100703] usb 1-9.1.4: New USB device found, idVendor=239a, idProduct=0078, bcdDevice= 1.00
[  +0.000015] usb 1-9.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000006] usb 1-9.1.4: Product: CUSTOM
[  +0.000006] usb 1-9.1.4: Manufacturer: NXP
[  +0.000004] usb 1-9.1.4: SerialNumber: DA081561D7091730B500005002004100
[  +0.002528] usb-storage 1-9.1.4:1.0: USB Mass Storage device detected
[  +0.000397] scsi host30: usb-storage 1-9.1.4:1.0
[  +0.001677] hid-generic 0003:239A:0078.00D2: hiddev1,hidraw1: USB HID v1.11 Device [NXP CUSTOM] on usb-0000:00:14.0-9.1.4/input1
[  +1.023484] scsi 30:0:0:0: Direct-Access     Adafruit UF2 Bootloader   1.0  PQ: 0 ANSI: 2
[  +0.000696] sd 30:0:0:0: Attached scsi generic sg7 type 0
[  +0.000431] sd 30:0:0:0: [sdh] 65801 512-byte logical blocks: (33.7 MB/32.1 MiB)
[  +0.000464] sd 30:0:0:0: [sdh] Write Protect is off
[  +0.000006] sd 30:0:0:0: [sdh] Mode Sense: 03 00 00 00
[  +0.000449] sd 30:0:0:0: [sdh] No Caching mode page found
[  +0.000004] sd 30:0:0:0: [sdh] Assuming drive cache: write through
[  +0.001728]  sdh:
[  +0.000123] sd 30:0:0:0: [sdh] Attached SCSI removable disk
```

hathach, 2023-03-09:
as suspected, this is host caching, it doesn't bother to re-read any of the descriptors.

closed 2023-03-09 by hathach