adafruit/tinyuf2 #313: Drive not appearing on ESP32-S3-DevKitC-N32R8

ishahak, 2023-04-21:
### Operating System

MacOS

### INFO_UF2.TXT

-

### What happened ?

When connecting to Mac or Windows, a drive is not appearing. The expected behavior is that after flashing for the UART connector, then connecting to the USB connector will show a mapped drive.

A full description appears here: https://forums.adafruit.com/viewtopic.php?t=200794

### How to reproduce ?

build, flash, connect. all described here: https://forums.adafruit.com/viewtopic.php?t=200794

ishahak, 2023-04-23:
This bug was reported from IDF v4.4.3 on Mac.
I did a reinstall of everything on a fresh Ubuntu and got it working as expected.
so this bug can be closed. thank you!

IMO you should add a comment here:https://github.com/adafruit/tinyuf2/blob/master/ports/espressif/README.md
saying that TinyUF2 is compatible with IDF 4, not 5

hathach, 2023-04-24:
you should use the idf with the same hash commit in the lib/

closed 2023-04-24 by hathach