adafruit/tinyuf2 #322: CURRENT.UF2 always blank

zvecr, 2023-06-05:
### Operating System

Windows 11

### INFO_UF2.TXT

TinyUF2 Bootloader 0.14.0-10-g86a054f - st/cmsis_device_f3 (v2.3.6) st/stm32f3xx_hal_driver (v1.5.6-9-g75cf4fc) tinyusb (0.15.0-331-ge3b3229d6)
Model: ST STM32F303 Discovery
Board-ID: STM32F303 discovery
Date: Jun  5 2023

### What happened ?

Content of `CURRENT.UF2` is always just zeros, and does not reflect the current flash contents,
```
hexdump ~/Desktop/CURRENT.UF2
0000000 0000 0000 0000 0000 0000 0000 0000 0000
*
0080000
```
(Where hexdump truncates identical lines as `*`)

This has been reproduced on both a `STM32F303 discovery` and `stm32f401_blackpill`, so the issue is not isolated to a single port.

### How to reproduce ?

Reboot to bootloader
Copy `.uf2` to device
Reboot to bootloader again
Copy `CURRENT.UF2` to host
Inspect content

MiyukiDark, 2023-11-01:
This issue will be on all MCUs where the flash did not start at address 0
It just calculates it totally wrong 
The issue is in ghostfat.c on lines 486 and on 
```
// CURRENT.UF2: generate data on-the-fly
      uint32_t addr = BOARD_FLASH_APP_START + (fileRelativeSector * UF2_FIRMWARE_BYTES_PER_SECTOR);
      if ( addr < _flash_size ) // TODO abstract this out
      {
```
addr is always bigger than the size of a flash

hathach, 2023-12-15:
should be fixed by #367, pleaes check it out

PR #367 (2023-12-15): Fix 322
## Description of Change

- add BOARD_FLASH_ADDR_ZERO, fix CURRENT.UF2 always blank for stm32 mcu. fix #322 
- add cmake for stm32f411 discovery board
closed 2023-12-15 by hathach