adafruit/tinyuf2 #358: Build for Adafruit Qualia S3 board

tyeth, 2023-11-13:
Hey there!
There doesn't seem to be an Adafruit Qualia build, the product item is 5800 (Adafruit Qualia ESP32-S3 for TTL RGB-666 Displays) https://www.adafruit.com/product/5800
One to add to the list I guess :)

tyeth, 2023-11-13:
Having just flashed the factory reset bin file then double reset into uf2 bootloader it's shown up as the adafruit tft experimenter when Windows installed the device.
I was originally trying to install circuitpython on the qualia with the web-installer method which failed looking for a tiny uf2 qualia release zip

hathach, 2023-11-14:
our bad, tftexperimental is prototype name of the board, we forgot to change it to qualia. PR #359 will address this, we will also bump up release after merged as well. Thank you for your issue.

closed 2023-11-14 by hathach

hathach, 2023-11-14:
PS: it is now available in release 0.18.1

tyeth, 2023-11-14:
Thanks @hathach!

PR #359 (2023-11-14): rename tft experimental to qualia s3
## Description of Change

- rename tft experimental (protoype) to qualia s3 (production) name. fix #358 
- update double tap pin from 5 to 4, remove neopixel since production has none