adafruit/tinyuf2 #370: Can't build Espressif boards on macOS with latest master.

UnexpectedMaker, 2023-12-24:
### Operating System

MacOS

### INFO_UF2.TXT

I don't have one - can't even build.

### What happened ?

ESP-IDF was removed from lib/esp-idf, but build script is still looking for it and failing when it's not there.

`error: pathspec '/Users/seon/tinyuf2/ports/../lib/esp-idf' did not match any file(s) known to git`

Reference here I think.
https://github.com/adafruit/tinyuf2/blob/fd8b95078ff1249c863a56ec017766921d8ae087/ports/espressif/CMakeLists.txt#L19

Even if I create a symbolic link from my esp-idf folder to lib/esp-idf, it still fails with the same error.

Maybe I'm missing a step - but I can't find any instructions that tap about how to include or reference the IDF now that it's been moved to 5.1.1 and removed as a submodule from lib.

### How to reproduce ?

clone a fresh repo, try to build.

UnexpectedMaker, 2024-01-04:
Hey @hathach - Happy new year :)
Any idea what's going on here?  I have new boards I want to create a PR for, but I want to test them first, and I am unable to build due to this issue.
Thanks

hathach, 2024-01-09:
should be fixed, please pull and try again.

closed 2024-01-09 by hathach

UnexpectedMaker, 2024-01-09:
Thanks! I'll give it a whirl today :)